dhalperi/linux-80211n-csitool-supplementary #75: How can I get CSI measurements on multiple links by capture beacon?

xuekevin, 2015-07-10:
I have read FAQ 5 and 11, but I want make sure if my thought can work out. my thought is the aps can broadcast beacon frames periodically. my laptop can capture these frames without associate to the ap. because I  can get the namelist of the ap which is provided by the beacon frame. then if the beacon is tranmitted at HT rate, I can get CSI message from it. which can achieve the goal to connect muliple aps in client/ap mode. Is my thought correct? because I read some paper, they all mention beacon to get CSI message. but how? It make me confused.
In FAQ 11, you recommend to use injection mode for most multi-link measurements, which means I can use three laptop, two as transmitter, one as receiver, and one receiver can receive two transmitters' simultaneously. how does it happen? does it about beacon? thanks!!!

dhalperi, 2015-07-10:
As in FAQ 5, the client must be connected to the AP. A client can only connect to a single AP at a time.

closed 2015-07-10 by dhalperi

xuekevin, 2015-07-10:
all right! I will try to injection mode the LORCON address does't work, can you give me another links? 
[image]

xuekevin, 2015-07-10:
and you say the  injection mode can achieve most multi-link measurements. how does it happen? can you give me a brief explaination. It will save my day!

xuekevin, 2015-07-10:
thanks! I have already downloaded LORCON!