dhalperi/linux-80211n-csitool-supplementary #76: Try injection mode encouter an error

xuekevin, 2015-07-10:


xuekevin, 2015-07-13:
@linkin8834,have you ever met this problem? thanks!

linkin8834, 2015-07-13:
Hi Kevin,
     I think you are not correctly using the driver. Please execute the following 
     ls /lib/firmware/ | grep iwlwifi
     and let me know what's the results

linkin8834, 2015-07-13:
Also, did you follow the instructions to build the modified firmware into your linux?

xuekevin, 2015-07-13:
Here is the results
iwlwifi-1000-5.ucode
iwlwifi-100-5.ucode
iwlwifi-105-6.ucode
iwlwifi-135-6.ucode
iwlwifi-2000-6.ucode
iwlwifi-2030-6.ucode
iwlwifi-3945-2.ucode
iwlwifi-4965-2.ucode
iwlwifi-5000-2.ucode
iwlwifi-5000-2.ucode.sigcomm2010
iwlwifi-5000-5.ucode.orig
iwlwifi-5150-2.ucode
iwlwifi-6000-4.ucode
iwlwifi-6000g2a-5.ucode
iwlwifi-6000g2a-6.ucode
iwlwifi-6000g2b-6.ucode
iwlwifi-6050-5.ucode

xuekevin, 2015-07-13:
I use the new installation instructions. my ubuntu is 12.04

linkin8834, 2015-07-13:
Hi Kevin,
    I have not encountered this problem before. I am using ubuntu 14.04. But before that I was using 12.04 and it is working.

xuekevin, 2015-07-13:
Is the results normal ? and what else should I do besides to follow tips as follows? could I miss some things?  thanks a lot! 
[image]
this step make install here, the directory is " linux-80211n-csitool-supplementary/injection/"  under this directory I just excute make order ,become make install can not work. Is the mormal?
[image]
by the way, could you give my your lorconv1 ? I want to make sure is not the library's problem.
my email is xuekaiwen_kevin@126.com

xuekevin, 2015-07-13:
@dpward , 
Hi David! Have you ever met my problem？what kind of problem can it be?

linkin8834, 2015-07-13:
Xue,
    Whats your email address? I can send you the lorcon code

xuekevin, 2015-07-13:
@linkin8834 , my email is xuekaiwen_kevin@126.com， you really help me a lot! thanks!

co12, 2015-07-13:
@linkin8834 , can you also send me a copy of lorconv1? I can not download it, my email is xushxin1991@163.com. thanks a lot!

xuekevin, 2015-07-13:
@linkin8834 @dpward  Sorry for asking again! I reinstall the ubuntu, when I try to git checkout the tag following instructions.
[image]
I met an error as follows:
[image]
the ${CSITOOL_KERNEL_TAG} is csitool-3.5 
Does it means the ubuntu version is not correct? Can you tell me what is your ubuntu version
My version is Ubuntu 12.04.2 LTS, the release version get by uname -r  is : 3.5.0-23-generic

hmlsi, 2015-07-13:
Hi, I am also facing error but this is at very initial stage. when i tried to install libpcap-dev using sudo apt-get install libpcap-dev
It gave me some very long error list. The main thing was " Package octave is not configured yet". I have already installed octave 4.0.0 and its working fine. Can anybody please help me?

linkin8834, 2015-07-13:
@co12  Sent. The 80211ninja which used to host the old lorcon code is down.

linkin8834, 2015-07-13:
@xuekevin I am getting 3.16.0-30-generic as the kernel version.

hmlsi, 2015-07-13:
@dpward  @linkin8834  can you please any help on this?

dpward, 2015-07-13:
The issue is very likely that the script is trying to reload the iwlwifi module with the parameter `debug=0x40000` but the kernel (and the modified driver) were built with `CONFIG_IWLWIFI_DEBUG=n`. The quick workaround is remove that parameter from the script.

ozawali, 2015-07-14:
@linkin8834  Hello, that I cannot download the  lorconv1 and can you send me this ? thank you!  my email is ozawa1314@gmail.com

linkin8834, 2015-07-14:
@ozawali  Sent.

ozawali, 2015-07-14:
@linkin8834   I don't know the  lorconv1 very well and what does this component function?

linkin8834, 2015-07-14:
@ozawali  It is used in injection mode for the binary file random_packets.

xuekevin, 2015-07-14:
@dpward thanks a lot ! you mean I should delete "modprobe iwlwifi debug = 0x40000", is that normal will It cause some other problems?

xuekevin, 2015-07-14:
@linkin8834 , @dpward  except for the step below,what else should I take care, do I need to set the address which is descibed in FAQ5 "The packet is addressed both to and from a hardcoded, fixed address"?

ozawali, 2015-07-14:
@linkin8834  Do I need install the libpcap-dev?

linkin8834, 2015-07-14:
@ozawali I didn't remember. I think it was installed by default

linkin8834, 2015-07-14:
@xuekevin  As far as I am concerned, I did not change that address.

xuekevin, 2015-07-14:
@linkin8834 thanks! After I remove the debug=0x40000 it works!

dhalperi, 2015-07-14:
Glad it's working!

closed 2015-07-14 by dhalperi

lixiaohai, 2015-11-09:
@linkin8834 Hello, that I cannot download the lorconv1 and could you send me this ? thank you! my email is lxh4217976@163.com

linkin8834, 2015-11-10:
@lixiaohai  You can try this link: https://github.com/dhalperi/lorcon-old

lixiaohai, 2015-11-11:
@linkin8834 I have downed it.Tks!