dhalperi/linux-80211n-csitool-supplementary #77: Injection mode issue

hmlsi, 2015-07-14:
Hi, I am also facing error but this is at very initial stage. when i tried to install libpcap-dev using sudo apt-get install libpcap-dev
It gave me some very long error list. The main thing was " Package octave is not configured yet". I have already installed octave 4.0.0 and its working fine. Can anybody please help me?

dpward, 2015-07-14:
Hmmm... try `dpkg-reconfigure octave` ?

dpward, 2015-07-14:
Also try:
`sudo apt-get clean`

hmlsi, 2015-07-15:
After running these commands, computer says

"/usr/sbin/dpkg-reconfigure: octave is broken or not fully installed"
I uninstall the octave by running "sudo apt-get remove octave" and all its system wide configuration files using "sudo apt-get --purge autoremove". Then again installed octave using 
sudo apt-add-repository ppa:octave/stable
sudo apt-get update
sudo apt-get install octave

Now this is what I am seeing at the end:

Setting up octave-common (3.8.2-4) ...
Setting up octave (3.8.2-4) ...
error: couldn't read directory /usr/local/share/octave/packages: No such file or directory
error: called from
    rebuild at line 29 column 7
    pkg at line 505 column 25
dpkg: error processing package octave (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for libc-bin (2.21-0ubuntu4) ...
Errors were encountered while processing:
 octave
E: Sub-process /usr/bin/dpkg returned an error code (1)

What to do? I have googled but could not find any appreciable solution

dpward, 2015-07-15:
It sounds like there is an issue with the package in the PPA. I would suggest you use the Octave package in the main Ubuntu repository (universe).

Remove the existing packages and the PPA:
`
sudo apt-get remove --purge '*octave*'
sudo apt-add-repository --remove ppa:octave/stable
sudo apt-get clean
`

Then try again.

hmlsi, 2015-09-09:
Hi,
While setting up injection mode, we are using channel 64. My question is, can we use any other channel (like channel 10 if I want to use 2.4 GHz band) ?

dhalperi, 2015-09-20:
Yes.

closed 2015-09-20 by dhalperi