dhalperi/linux-80211n-csitool-supplementary #78: download LORCONv1 from http://802.11ninja.net/svn/lorcon/branch/lorcon-old/

hmlsi, 2015-07-14:
This download link is not working.This is a continuous issue which almost everyone is facing. Can someone provide any alternative?

ozawali, 2015-07-14:
you can download from the IEEE802.11

dhalperi, 2015-07-14:
@hmlsi I sent a request to the authors of lorcon to see if they can post a new version of it. In the meantime, would anyone who has a working copy of lorcon-old care to put a mirror on github?

dhalperi, 2015-07-14:
@ozawali I don't think that's correct.

co12, 2015-07-15:
@hmlsi ， I don't know how to put a mirror on github, but I can send you a working copy by email

linkin8834, 2015-07-15:
I notice that as well. I keep a local copy at my disk.

hmlsi, 2015-07-15:
Thanks a lot all of you. My email id is hmlsi@yahoo.com

co12, 2015-07-15:
@hmlsi Already sent, please check

dhalperi, 2015-07-15:
If you send me a copy (daniel@halper.in) I will create a mirror on github and update the instructions. Thanks!

dhalperi, 2015-07-16:
Mirror is up here: https://github.com/dhalperi/lorcon-old

closed 2015-07-16 by dhalperi

hmlsi, 2015-07-16:
Thank you all of you :)