dhalperi/linux-80211n-csitool-supplementary #79: No CSI messages received in injection mode?

xuekevin, 2015-07-14:
@dpward  
Hi David!
After I remove the parameter *debug=0x40000* , It seems become normal! everytime I when I execute **./setup_inject.sh 64 HT20**, the ubuntu system will  turn off the internet, then reconnect. I think it is  abnormal. So I disconnect the internet manually. But int receiver side they did not get any packets. Here is the operations I did, have I missed something? Is the output normal ? thanks!
* transmitter side
[image]
* receiver side 
[image]
receiver side have nothing get. both transmitter and receiver side are not connected with internet (AP).

xuekevin, 2015-07-14:
@linkin8834,  Hi!  you said before that "I am getting 3.16.0-30-generic as the kernel version." could you give me you system's download link, Is it the 32bit or 64bit?  Thanks!

co12, 2015-07-14:
@xuekevin , with the help of linkin8834, I solved the problem just now. 

I suggest before you use the inject mode, you should disable the network manager using the following line
sudo /etc/init.d/network-manager stop

linkin8834, 2015-07-14:
@xuekevin Please disable the network manager first and make sure you modprobe correctly :}

dhalperi, 2015-07-14:
+1 thanks @linkin8834

closed 2015-07-14 by dhalperi

xuekevin, 2015-07-15:
@linkin8834 @dhalperi  @co12  thanks for all your suggestions. but It still doesn't work.after I execute 
** ./setup_inject.sh 64 HT20**, I execute **iwconfig** the result is as follows.
[image]
wlan0 's mode is Managed,  I think it should be Monitor
the different with your script in the **setup_inject.sh** is I remove debug=0x40000 because the error met before **FATAL： Error inserting iwlwifi (lib/modules/3.5.0-23-generic/updates/iwlwifi.ko): Invalid argument**
[image]

So the problem is the wlan0's mode should be Monitor, but it is not. what tips should I do to solve this problem? thanks all your guys!

linkin8834, 2015-07-15:
backup /etc/network/interfaces and then put these into this file

auto lo
iface lo inet loopback
iface wlan0 inet manual

then reboot. Turn off the network manager

xuekevin, 2015-07-15:
@linkin8834 , I tried, the wlan0's mode still is managed. so the normal conditions the mode should be monitor? and following tips in the install instructions should I take ?
[image]

xuekevin, 2015-07-15:
@linkin8834  by the way, Is you ubuntu system 14.04 and 64 bit or 32 bit?

linkin8834, 2015-07-15:
@xuekevin 14.04 64bit

xuekevin, 2015-07-15:
@linkin8834. thanks a million! so the normal result is the wlan0 should be monitor mode?

linkin8834, 2015-07-15:
@xuekevin  Need to double check on that when I am back at lab :D

xuekevin, 2015-07-15:
@linkin8834 Thanks! Wish you have a good day!

xuekevin, 2015-07-16:
@linkin8834 @dhalperi  @dpward  thanks you all ! 
Eventually the injection mode can work. I reinstall my ubuntu system.
**ubuntu  version**：14.04 LTS  32bit
**the kernel tag** ： 3.16.0.30-generic
  I followed the new Installation Instructions 
and when I execute ./setup.inject.sh I met error as @linkin8834  asked before
`<command failed: Device or resource busy (-16)>` so I used @linkin8834 's script. It works!

linkin8834, 2015-07-16:
@xuekevin Even using the script you might still see that **command failed** stuff from time to time..Just try to execute the script several times until this message disappears, or just reboot your system.

xuekevin, 2015-07-17:
@linkin8834 ， Yes. I have noticed that problem. thanks!

xuekevin, 2015-07-17:
@linkin8834  by the way,  I found the number of the tramsmitter PC's antenna is 1, I want to make it to be 2 or 3. Do you know how to implement this? and `/random_packets 1 100 1` if I change the packets size to 500, what it is will be affected?

linkin8834, 2015-07-17:
@xuekevin  No. You need to change the rate according to FAQ #1.

co12, 2015-07-17:
Refer to FAQ#1, if you want to use 3 antennas to send packets in a higher rate, you can run

echo 0x1C113 | sudo tee `find /sys/ -name monitor_tx_rate`

xuekevin, 2015-07-17:
thanks ! let me have a try! @co12

co12, 2015-07-17:
@xuekevin , Can you connect to an unencrpted AP to get packets from commercial AP?

linkin8834, 2015-07-17:
@co12 Manually connect to that AP first, and try to PING it. Does PING goes through? If yes then you have high probability that this can work. Try to ping with small interval, say, 0.01, to 'force' 802.11n rate (not confirmed to be true, but works for me). Do not use network manager to automate this process which is not that stable.

co12, 2015-07-17:
Thank you for your reply, I disable network-manager, use the ifconfig /iwconfig /dhclient to connect the AP manually, it works.

It seems like that when to get csi, network-manager should be disabled and manually to  set wireless network.

2015-07-17 

xushxin1991

xuekevin, 2015-07-21:
@co12  I just see this ,sorry about that! I once use AP mode to get packets from commerical AP. It worked.

lidongning, 2017-06-21:
I have the same problem ,  I have wlan0 : not associate  mon0: 5Ghz  but when  injection ,it still shows Generatting....payloads     how  to set ???

lidongning, 2017-06-21:
I have the same problem , I have wlan0 : not associate mon0: 5Ghz but when injection ,it still shows Generatting....payloads how to set ???   @xuekevin   @linkin8834