dhalperi/linux-80211n-csitool-supplementary #80: some questions about the spatial streams

ozawali, 2015-07-15:
Hi,@dhalperi   Thank you for you recommendation that reading you PH.D thesis and I have some questions than maybe you can help me. first,I cannot understand the spatial streams very well,those streams are from which layer and those streams how to come into being. In the paper,  spatial streams are present before that antennas transmit the signals.

dhalperi, 2015-07-15:
Do you have a specific question? I'm confused by your post.

dhalperi, 2015-07-15:
You may also wish to read "802.11 with multiple antennas for dummies", a simple overview article we wrote for CCR. http://www.sigcomm.org/node/2821

ozawali, 2015-07-15:
Thank you I will and I have read the chapter 1 to 4 that give me many information.  Thank you very much

closed 2015-08-10 by dhalperi