dhalperi/linux-80211n-csitool-supplementary #82: the question about the Linux kernel

ozawali, 2015-07-16:
@dhalperi Hello, I am confused that why the  kernel need to re-compile and I want to know the modified part of the custom Linux kernel you provide since the later we will install the custom firmware of the INC 5300 that you modified for getting the CSI. Thank you !

dpward, 2015-07-16:
The custom firmware sends beamforming messages (CSI), but for it to be useful, there has to be a way to read and store these messages. The `iwlwifi` driver (for the Intel adapter) processes the beamforming messages and delivers the CSI data to userspace. This is the only part of the kernel that is changed.

dhalperi, 2015-07-16:
If you would like to see the changes, a typical way to do this is as follows:

1. See the kernel history with `git log`
2. Find the most recent commit that is NOT authored by @dpward or @dhalperi. This is typically a commit by Linus Torvalds. Record its commit hash (`<HASH>`).
3. Run `git diff <HASH>` to see what changes have been made between our custom kernel and the stock Linux kernel.

Then you can review all changes.

Here is a manually-constructed link on GitHub, but it does not always work. Because the Linux kernel is so big, is sometimes it takes too long to load, and GitHub might time out loading the page.

dhalperi, 2015-07-16:
https://github.com/dhalperi/linux-80211n-csitool/compare/39a8804455fb23f09157341d3ba7db6d7ae6ee76...62b04a17f6b1208a89b1c0ed055f02378a4130eb

ozawali, 2015-07-16:
@dpward  Thanks！ I also want to know that the custom firmware you provide if have modified by you and if I just use the NIC5300 connect to the commercial AP ,the beamforming message are from the AP since we use the receiver-side computation that's right？

ozawali, 2015-07-16:
@dhalperi  Thank you！ I have seen the part of the differences

ozawali, 2015-07-17:
@dhalperi  In addition, since I have read the 802.11API and your PHD thesis ,I have another question that the CSI information compute at the receiver-side(e.g client model with 5300) then the CSI information how to send to the transmitter-side by explicit feedback? from my point of view that CSI information carride by a kind of the frame that 802.11 included.

ozawali, 2015-07-28:
and also the Algorithm that you developed have been written to the NIC?

dhalperi, 2015-08-10:
The CSI is not sent back to the transmitter. It is instead sent up to userspace. The message sent to userspace is the type that would be sent back to transmitter over the air.

closed 2015-08-10 by dhalperi

y119777, 2015-08-11:
@ozawali ，i want to konw how the csi is constructed,could you give me some advise