dhalperi/linux-80211n-csitool-supplementary #84: How to use only one Tx antenna

eric11220, 2015-07-27:
We used MacAir as AP and a Fujitsu as receiver with network adapter switched to intel5300 with the modified driver provided.

As far as I know MacAir has 2 antennas, so the max number of Tx antennas is 2, Rx 3
However, we found that the size of csi we received is sometimes 1(Tx)*3(Rx)*30 and other time 2(Tx)*3(Rx)*30.

So my question is that is it possible to force Tx to only use 1 antenna instead of drifting between 1 and 2 ? 

If not, is it possible to know which antenna Tx is using when csi size is 1(Tx)*3(Rx)*30 from the receiver side?

ozawali, 2015-07-28:
Yes, you can set the rate and the number of the TR antennas on the injection mode!

dpward, 2015-07-28:
Just to be clear, your MacBook Air does not have an Intel 5300 adapter and is not running Linux — just the Fujitsu system — correct? If that is the case, no, you can't set the number of antennas (or transmit rate) on the MacBook Air.

While there's not any way of knowing which antenna your transmitter used by just looking at the receiver collected CSI, depending on adapter/driver support in the MacBook, it might be possible to use Wireshark in monitor mode there and capture outgoing packets, which might indicate the transmit antennas used.

eric11220, 2015-07-28:
Thanks! 
Really appreciate the quick and informative responses! :)

eric11220, 2015-07-29:
We tried WireShark on our MacBook Air and did see header radiotap.antenna (index of antenna packet is transmitted or received over, starting at 0). 
However, we found that it's always 0 even when the receiver gets 2(Tx)*3(Rx)*30 CSI.

Also, we found that channel type is always 802.11g in WireShark, which made us confused, since as far as we understood, only 802.11n HT packet would be analyzed for CSI? 

Please do correct me if I misunderstood some part!
Thanks in advance!

dhalperi, 2015-08-10:
I would not trust the radiotap headers; use the information available in MATLAB.

closed 2015-08-10 by dhalperi