dhalperi/linux-80211n-csitool-supplementary #85: Setting up transmitting rate

hmlsi, 2015-07-27:
I have read FAQ#1. I am using commercially available TP link. using this as AP, can I set my rate?

ozawali, 2015-07-28:
No , you must on the injection mode.

dpward, 2015-07-28:
@ozawali injection mode is not required for setting the rate.

@hmlsi You can set the transmit rate from a client with an Intel 5300 adapter to a commercial AP (which doesn't provide CSI, as the AP is receiving). However you can't set the transmit rate on a commercial AP. You can set the rate though with an Intel 5300 adapter operating as an AP.

hmlsi, 2015-07-28:
Got it. That means for setting transmit rate, transmitter must have Intel 5300. Thanks

ozawali, 2015-07-28:
@dpward  I mean you can set the rate and the antennas of the transmitter  on the injection mode, but for the csi data  although you change the commercial ap rate but the rate of getting csi is not useful

closed 2015-08-03 by hmlsi