dhalperi/linux-80211n-csitool-supplementary #86: wlan0 not found

muneebr1, 2015-07-27:
Hi, 
I am trying to set a platform for csi tool. I am following updated instructions. Everything worked fine till step-4. After that I disabled network manager using
 "sudo service network-manager stop".
 Now, I am unable to connect to any wireless network using iw command. Later, I restarted network manager using
 "sudo service network-manager restart".
 I am able to connect to LAN but no wifi device are shown.
 ifconfig -a is not showing wlan0. I have already tried "rfkill unblock wifi", ifconfig wlan0 up (its output is: wlan0:error while getting interface flags:no such device). output of "sudo ifup wlan0" : ignoring unknown interface wlan0=wlan0.I have unloaded and then reloaded wifi drivers and then tried these commands. 
I know this is not csi related issue. I am new to ubuntu and don't know what to do. Any kind of help is highly appreciated.

ozawali, 2015-07-28:
you can check out use the command "ifconfig" then show the wlan0 or wlan1 etc.

dpward, 2015-07-28:
Since the wlan0 interface is no longer appearing, I assume you ran the optional commands to prevent the wireless driver from loading during boot (very end of step 1). If you follow that method, then you have to load the driver manually when you first boot (`sudo modprobe iwlwifi`).

Or you can undo this step by deleting this file you created:
```
sudo rm /etc/modprobe.d/csitool.conf
```

muneebr1, 2015-07-28:
i did not run the optional commands . I only ran sudo service network-manager stop and after that wlan0 is not appearing. Ifconfig is showing lo and eth0.

ozawali, 2015-07-28:
That mean your NIC cannot have the driver

muneebr1, 2015-07-28:
I have re installed ubunto, and will do everything from scratch. Kindly tell me the command that i should use for disabling the network manager and also the command required for enabling the network manager once that i have installed the modified drivers.

dpward, 2015-07-28:
First, make sure you have read FAQ #7: with the modified driver you will be unable to connect to encrypted APs (using WPA/WPA2, etc.) However you should still see the wlan0 interface on your system as long as either the original driver or the modified driver is loaded — even if NetworkManager is not running, or if there is not a connection to a wireless network.

It's not actually necessary to stop NetworkManager during the installation if you intend to use it with the modified driver. (However for collecting CSI in a controlled manner, we strongly recommend you configure the interface manually rather than rely on NetworkManager.) If you want configure the wireless interface manually instead, you need to disable NetworkManager's control over it. NetworkManager will otherwise still run on the other interfaces and allow you to connect to a wired network, for example. The commands you need are in step 1 of the installation instructions:

```
sudo apt-get install iw
echo iface wlan0 inet manual | sudo tee -a /etc/network/interfaces
sudo restart network-manager
```

dhalperi, 2015-08-10:
Closing for inactivity

closed 2015-08-10 by dhalperi