dhalperi/linux-80211n-csitool-supplementary #88: injection mode help

muneebr1, 2015-07-29:
Can someone please explain following things
1. what is actually injection mode? I have read FAQ#5 but I am not clear yet. Can you provide me some reading material? what is mode 1 in it etc
2. Can we use injection mode if network manager is not disabled on transmitter side? I am intending to use Sony vaio ultrabook T13 with NIC 5300 as a transmitter

ozawali, 2015-07-30:
The injection mode not like other ap,client mode. in this mode you need't to connect the TR and when the TR send the packet the receiver in the monitor mode can get the packet. it is the standard includede in the 802.11n.  so if you want get multipath once you can use this mode. as for csi , you can also get if you install the csitool provided by the author.

hmlsi, 2015-08-02:
@dpward , Sir, can you please put some light on this issue? I am also having difficulty in understanding injection mode working. 
1.Can someone provide any reading material?
2.How actually csi packets are received in injection mode? Do I need to ping my transmitter in injection mode or what? 
3.in injection mode, does transmitter need to be connected with any network?
After reading previous issues, I have got a point that I can use only injection mode if I want to receive csi packets from multiple transmitters. Am I right?
It will be really helpful if someone can provide any reading material or can elaborate these questions. These might be childish questions but I don't know anything about injection mode working and could not get enough by reading FAQ#5 and previous issues

Thanks

ozawali, 2015-08-03:
1.injection is included in 802.11n standrad
2.you neednot ping, it just a mode,so you need make configure step by step.
3.make sure you read FAQ1

dpward, 2015-08-03:
Sorry for the delay.

There are different operating modes for 802.11 wireless networks. The most commonly used is *infrastructure mode*, in which one station is an access point and the other stations are clients, and all communication happens between a client and the access point (there is no direct communication between clients; the access point serves as a relay in this case). Another is *ad-hoc mode*, where stations are each configured (in what is called an IBSS) to communicate to each other directly; there is no access point. More recently, there are also modes for mesh networking (802.11s) and Wi-Fi Direct, although I believe these aren't supported by the driver for this adapter. Once stations are connected through one of these modes, they can be configured to use IP (either manually, or using DHCP) and then they can exchange IP packets sent inside 802.11 data frames.

For an Intel 5300 adapter with the modified firmware and driver, when it is operating in these modes, it will collect a CSI sample for each data frame that it would normally receive from the connected network. This includes unicast frames addressed to that station, as well as multicast/broadcast frames for addresses to which the station is subscribed. It does not include frames addressed to other stations or sent on other 802.11 networks.

Aside from using one of the normal operating modes, this adapter can be placed in "monitor mode", where it observes 802.11 traffic on a channel without being connected to a network. This is not limited to data frames, but can also include beacons and other management frames. Tools such as Wireshark are able to capture and display frames that are received in monitor mode, which include the 802.11 link-layer header and other information provided by the driver.

In monitor mode, there is no IP connectivity to other stations. However, it is possible to "inject" custom frames to be transmitted by the adapter. These frames do not belong to a network, but are structured like other 802.11 data frames, having MAC addresses in the header, a payload, etc. Other stations that are placed in monitor mode can detect these frames.

In addition to collecting CSI on connected networks as described above, the modified firmware will collect CSI for any frame it detects in which the source and destination MAC addresses are 00:16:ea:12:34:56. The code in the supplementary material injects frames with these addresses, allowing CSI to be collected without being connected to a network. The advantage to this is that no connectivity is necessary for simply collecting CSI, and it seems to work around some erratic behavior of the firmware that can happen when trying to use it on a connected network. However, you're no longer operating in a connected network and exchanging IP packets, although you can customize the payload of injected packets if you wish.

Note, it is possible to simultaneously be connected to an 802.11 network and to *also* place the adapter in monitor mode, by creating a second interface for the adapter like this:
```
sudo iw phy phy0 interface add mon0 type monitor
sudo ip link set mon0 up
```
rather than changing the mode of the existing interface.

dpward, 2015-08-03:
@ozawali No, injecting packets in this way does not follow the 802.11 standard; the standard only defines communication using one of the defined operating modes. Injection is also not specific or new to 802.11n &mdash; injection can certainly be done on 802.11a/b/g networks (still subject to adapter/driver support).

ozawali, 2015-08-03:
@dpward   Thanks for correct me!

hmlsi, 2015-08-03:
Thank you @ozawali  and thanks a lot @dpward . In your last sentences, what do you mean by changing the mode of existing firmware?

dpward, 2015-08-03:
By default, one interface is created for the adapter, named wlan0, which will be in client (infrastructure) mode. You can configure this interface to connect to a network that has an access point. In general (meaning without the CSI Tool), this is often done under Linux by using NetworkManager. If NetworkManager is not controlling the wlan0 interface, then the `iw` command can be used to connect to the network, and the `ip` or `dhclient` commands can assign the IP configuration on the interface.

Alternatively, you can change wlan0 from client mode to access point mode or ad-hoc mode. Ad-hoc mode can also be configured through either NetworkManager or command-line tools. Access point mode requires additional software (hostapd).

You can also change wlan0 to operate in monitor mode instead, where it will not have any connection but will observe traffic and can inject frames.

If you want to be connected to a network, but you simultaneously want to observe the traffic (including data frames to other stations, link layer headers, management frames, and beacons from other networks) you can add a second interface in the kernel that uses monitor mode (mon0), while the wlan0 interface remains in one of the normal operating modes.

hmlsi, 2015-08-04:
This is pretty much explanation. Thanks a lot

muneebr1, 2015-08-06:
Thank you so much for kind responses.
I have done these steps to set up injection mode
sudo apt-get install libpcap-dev
cd lorcon
./configure
make
sudo make install

cd linux-80211n-csitool-supplmentary/injection
make
sudo iw phy phy0 interface add mon0 type monitor
sudo ip link set mon0 up

and then on transmitter side: 

sudo echo 0x4101 |sudo tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/iwldvm/debug/monitor_tx_rate
sudo ./random_packets 1 100 1
 
on receiver side:
No csi packets are being captured

ISSUES:
1. No csi packets are collected on receiver side. Receiver is Dell inspiron n5110.
2. on transmitter, after sometime, mode of mon0 appears to be 
mode: managed. 
output of iwconfig:

eth0      no wireless extensions.

mon0      IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wlan0     IEEE 802.11abgn  ESSID:"LUMS-Guest"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: E0:5F:B9:36:AD:51   
          Bit Rate=86.7 Mb/s   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=42/70  Signal level=-68 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:5  Invalid misc:90   Missed beacon:0

lo        no wireless extensions.

Why it is so? Is it due to the fact that I am connected to the wifi at the same time?

dpward, 2015-08-06:
Oh, make sure NetworkManager isn't controlling the `mon0` interface (run the commands below once on each system, then try to repeat what you did above):

```
echo iface mon0 inet manual | sudo tee -a /etc/network/interfaces
sudo restart network-manager
```

That may explain why the interface is switching to managed mode.

I'm not sure that is all that is going on though. If the commands above don't help, then let's try removing the `wlan0` interface that you are using for Wi-Fi before adding the `mon0` interface. So run:

```
sudo iw dev wlan0 del
```
before you run
```
sudo iw phy phy0 interface add mon0 type monitor
sudo ip link set mon0 up
```

dhalperi, 2015-08-10:
@muneebr1 any luck?

muneebr1, 2015-08-10:
Thank you sir @dhalperi for asking. I followed following procedure
On Receiver:
sudo ip link set wlan0 down
sudo ip link set wlan0 type monitor
sudo ip link set wlan0 up
sudo iw wlan0 set channel 64 HT20
sudo ifconfig wlan0 up
sudo linux-80211n-csitool-supplementary/netlink/log_to_file log1.dat

On transmitter:
cd linux-80211n-csitool-supplementary/injection 
./setup_inject.sh 64 HT20
Output was:
"command failed: Device or resource busy (-16)"
iwconfig
eth0      no wireless extensions.

mon0      IEEE 802.11abgn  Mode:Monitor  Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wlan0     IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.

muneeb@Ultrabook:~$ sudo ifconfig wlan0 down
muneeb@Ultrabook:~$ iwconfig
eth0      no wireless extensions.

mon0      IEEE 802.11abgn  Mode:Monitor  Frequency:2.412 GHz  Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wlan0     IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.

muneeb@Ultrabook:~$ sudo iw mon0 set channel 64 HT20
muneeb@Ultrabook:~$ sudo ifconfig mon0 up
sudo echo 0x4101 |sudo tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/iwldvm/debug/monitor_tx_rate
sudo ./random_packets 5 100 1

After this point, Receiver was receiving csi packets

muneebr1, 2015-08-11:
Thank you sir @dhalperi for asking. I did following:

On Receiver:

sudo ip link set wlan0 down
sudo ip link set wlan0 type monitor
sudo ip link set wlan0 up
sudo iw wlan0 set channel 64 HT20
sudo ifconfig wlan0 up
sudo linux-80211n-csitool-supplementary/netlink/log_to_file log1.dat

On transmitter:
cd linux-80211n-csitool-supplementary/injection 
./setup_inject.sh 64 HT20
Output was:
"command failed: Device or resource busy (-16)"
output of iwconfig
eth0      no wireless extensions.

mon0      IEEE 802.11abgn  Mode:Monitor  Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wlan0     IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.

muneeb@Ultrabook:~$ sudo ifconfig wlan0 down
muneeb@Ultrabook:~$ sudo iw mon0 set channel 64 HT20
muneeb@Ultrabook:~$ sudo ifconfig mon0 up
sudo echo 0x4101 |sudo tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/iwldvm/debug/monitor_tx_rate
sudo ./random_packets 5 100 1

After this point, Receiver was receiving csi packets.

One question: if I want to collect csi packets for a certain time interval (say 1 min) what should i do??

dpward, 2015-08-11:
One way to achieve this is to introduce a fixed delay between transmissions; then you can choose the number of packets to transmit based on this delay and your desired time interval for collecting CSI. You can add a fourth parameter to the `random_packets` command to specify the delay in microseconds, and then adjust the first parameter with the number of packets you want to transmit.

An alternative (for either the transmitter or receiver) is to use the shell to start the process in the background, wait for a certain time, and then terminate the process. For example:

```
sudo sh -c 'linux-80211n-csitool-supplementary/netlink/log_to_file log1.dat & sleep 60; kill -TERM $!'
```

dpward, 2015-08-11:
Also, it's not necessary to run `sudo ifconfig wlan0 up` after running `sudo ip link set wlan0 up`, because these two commands do the same thing. (`ip` is intended to replace `ifconfig`.)

lixiaohai, 2015-11-11:
@muneebr1 

Hello!I do as you said on each system,but still can't receive any csi packets.Could you show your ifconfig and iwconfig results on both sides?What's more,could you be kind to show  your detail steps from the beginning to the end and the screen when it's working normally on both sides?
Thank you!Look forward to your favourable reply!

lixiaohai, 2015-11-16:
@muneebr1 

Hello!I re-install the ubuntu14.04 and do as you said.Finally,i can receive csi packet!Thank  you!

dhalperi, 2015-11-21:
Thanks for the update!

closed 2015-11-21 by dhalperi

simondchen, 2016-04-09:
Dear @muneebr1  I am using monitor mode, and I follow the https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection, but I can't get any csi data. I can get csi data in the AP mode, but the AP mode is not stable, so I change to monitor mode, but no csi data captured. I'm using ubuntu 11.04, is this version too old, can you show your detailed steps, thanks a lot

simondchen, 2016-04-09:
@muneebr1 solved! thanks

lingyan0222, 2016-05-26:
@dhalperi 

The issue is that collecting 40MHz csi data, which **monitor_tx_rate** to set? **0x4101** or **0x4801**?

the receiver:
    **./setup_monitor_csi.sh 64 HT40-**

the transmitter 
    **./setup_inject.sh 64 HT40-**

lingyan0222, 2016-06-04:
Hi @dhalperi @dpward 

Using injection mode to collect 40MHz CSI data form 5GHz, the channel parameters confused me that  HT64+ or HT64- for the CSI signal form 60-64 channel bonding for 40M GHz.

for example, ./set_injection,sh 64 HT40+, this is the primary channel is 64? In this command, it is HT40+ that the secondary channel 60 is above the 64?

Thank you a billion!

asifhanif1122, 2016-06-04:
@lingyan0222  Run this command **_iwlist channel_** to see which channels are supported by NIC. HT40+ means next higher adjacent channel will be merged with primary channel. 64 is your primary channel. See following fig.  HT40+ would not work here because next channel is 100 and there is a jump from 64 to 100. But you can use HT40- as adjacent channel back to 64 is 60. 
[image]

lingyan0222, 2016-06-06:
@asifhanif1122  Thank you for your feedback. 
Anther issue is that which is the carrier frequency and the center frequency of 40MHz, after this channel bonding?

elegance0222, 2016-06-06:
fsg

asifhanif1122, 2016-06-06:
@elegance0222  Here "Below" means smaller value in frequency . Channel 100 is not below in this case. 64 HT40- will take channel 64 as primary channel and -ve sign will pick a channel that is 20MHz far in backward direction as secondary channel. And channel 60 will your secondary channel because its distance from channel 64 is 20MHz. 
Channel 64 ===> 5.32 GHz
Channel 60 ===> 5.30 GHz

Distance between Channel 64 & Channel 60 = 5.32GHz - 5.30GHz  = 20 MHz

elegance0222, 2016-06-07:
@asifhanif1122  Thank you again.
 
The channel 60 and 64 bonding for 40MHz, the central frequency is the mean of both central frequency. In this case, it is 5.31GHz?

asifhanif1122, 2016-06-07:
@elegance0222  Yes. It will be mean of two central frequencies. 5.31GHz is correct in this particular case.

elegance0222, 2016-09-23:
@dhalperi @asifhanif1122 
In order to get 40MHz (5GHz) CSI data in injection mode, which code to set 'monitor_tx_rate'? 0x4901, 0x4801, or 0x4101? I have find that  it maybe affect the phase of the CSI. 
Thank you.

asifhanif1122, 2016-09-24:
Please check my work on rate selection for CSI

[image]

ZhijingLi, 2017-01-24:
Hi,

I don't quite understand why the monitor mode (The packet is addressed both to and from a hardcoded, fixed address 00:16:ea:12:34:56). Does this mean I cannot get CSI when I do passive scan of surrounding devices?

saqib1707, 2017-12-16:
@dpward  can we get CSI value from router ? for example a transmitter sending packets which are received by router which then computes the CSI matrix and sends it to some pc/laptop so that we get to see the CSI value between the transmitter and router

chen0408, 2018-07-14:
A lot of csi information is collected in monitor mode, so how can I determine which sender sent the csi information to be generated?

muhmaz3, 2019-08-15:
@asifhanif1122 
The figure you have posted shows 17 bits, but 0x4101, as you have said, should consist of 16 bits so which bit is extra here? Can you kindly share the source where these bits are explained?

culiver, 2020-03-17:
@simondchen I have encountered a similar situation as you had mentioned. Can you share how you solved that problem?