dhalperi/linux-80211n-csitool-supplementary #90: about the real time process

ozawali, 2015-08-03:
@dpward  @dhalperi Hello, I want to use the CSI  in real time, but maybe need the API for the userspace about get CSI data?  so I want ask whether I can do this job. if I can do this ,that Ishould know the API about the CSI data get function?

linkin8834, 2015-08-04:
Just a comment here: use MATLAB system function. http://www.mathworks.com/help/matlab/ref/system.html. 
So actually you use matlab to call the log_to_file and process the data in real-time with MATLAB.

dpward, 2015-08-04:
@ozawali Yes you can. The modified driver uses a netlink connector interface to send CSI samples into user space. `log_to_file` listens for messages on this netlink connector interface, and it writes the CSI contained in each message into a file. You can also write your own C program that uses this netlink connector interface to receive messages containing CSI and process them in real time.

The best starting point is the source code for `log_to_file`:
https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/netlink/log_to_file.c
Instead of calling `fwrite()` to store the CSI in a file, you can do something else with it.

You may also want to refer to the modifications to the iwlwifi driver to understand the different types of messages that can be sent using the netlink connector interface and their format:
https://github.com/dhalperi/linux-80211n-csitool/compare/39a8804...dhalperi:csitool-4.0#files_bucket

ozawali, 2015-08-05:
@dpward  Thank you for your advice

closed 2015-08-05 by ozawali

Gy286374, 2019-07-20:
Does anyone use Java language to achieve real-time access to CSI information? If so, can you tell me how to achieve it?