dhalperi/linux-80211n-csitool-supplementary #91: Load unmodified firmware without rebooting

linkin8834, 2015-08-04:
Hello,
       I am wondering if it is possible if I could reload the unmodified firmware without rebooting. Here is what I do currently:
       1. By default, I am using 'unmodified firmware'. Then, I run a script to copy the modified firmware into /lib/firmware/ and rename the unmodified one to something else, so that when rebooting, the unmodified one would not get loaded.
       2. Reboot
       3. Now, with the modified firmware, I can do CSI measurement. When I want to get WiFi back, I have to remove the modified firmware from /lib/firmware, and restore the unmodified one into that folder.
        4. Reboot
        5. I can go to Internet again.
But it takes me two reboot in the process... is it any way to avoid them? Thanks!

dpward, 2015-08-04:
Yes, instead of rebooting, you just need to unload and reload the iwlwifi module, for example:

```
sudo modprobe -r iwlwifi mac80211
sudo modprobe iwlwifi connector_log=0x1
```

linkin8834, 2015-08-04:
Hi David,
   Thank you. I will try tomorrow.

linkin8834, 2015-08-07:
Hi David,
    Here is what I tried today, when I want to switch back to 'Internet mode':
    1. Move back the original /lib/firmware/iwlwifi-5000-5.ucode back into /lib/firmware
    2. remove the symbolic link /lib/firmware/iwlwifi-5000-2.ucode
    3. modprobe -r iwlwifi mac80211
    4. modprobe iwlwifi
    But still, I cannot connect to any encrypted routers. I need a restart to get back to Internet. 
    Strangely enough, when I restart, it seems that it still uses  /lib/firmware/iwlwifi-5000-2.ucode as the driver. Here is the output of modinfo iwlwifi:
   $ modinfo iwlwifi
filename:       /lib/modules/3.13.0-24-generic/updates/iwlwifi.ko
license:        GPL
author:         Copyright(c) 2003-2013 Intel Corporation <ilw@linux.intel.com>
version:        in-tree:
description:    Intel(R) Wireless WiFi driver for Linux
firmware:       iwlwifi-100-5.ucode
firmware:       iwlwifi-1000-5.ucode
firmware:       iwlwifi-135-6.ucode
firmware:       iwlwifi-105-6.ucode
firmware:       iwlwifi-2030-6.ucode
firmware:       iwlwifi-2000-6.ucode
firmware:       iwlwifi-5150-2.ucode
firmware:       iwlwifi-5000-2.ucode
firmware:       iwlwifi-6000g2b-6.ucode
firmware:       iwlwifi-6000g2a-5.ucode
firmware:       iwlwifi-6050-5.ucode
firmware:       iwlwifi-6000-4.ucode
firmware:       iwlwifi-3160-7.ucode
firmware:       iwlwifi-7260-7.ucode

dpward, 2015-08-07:
Actually, it's a bit misleading. The firwmare information you see with `modinfo` are strings that are compiled into the driver; it doesn't show what firmware is actually loaded by the driver. (You can run `modinfo` even if the iwlwifi module is not loaded and get the same output.)

To see what firmware is loaded, check `dmesg` and search for "iwlwifi [...] loaded firmware version". The unmodified version that is distributed with Ubuntu is 8.83.5.1.

linkin8834, 2015-08-07:
Hi David,
    I checked the dmesg. The loaded firmware is 8.83.5.1. I follow your previous advice to remove the *.ko files from the update folder. Here is what I did just now:
     1. Move back the original /lib/firmware/iwlwifi-5000-5.ucode into /lib/firmware
     2. remove the symbolic link /lib/firmware/iwlwifi-5000-2.ucode
     3. remove the *.ko files in the **update** directory
     4. modprobe -r iwlwifi mac80211
     5. modprobe iwlwifi
     6. restart the network manager.
I can see that 8.83.5.1 is loaded in the dmesg after these steps. And also, modinfo tells me that kernel/drivers/net/wireless/iwlwifi/iwlwifi.ko is loaded. However, still, I cannot connect to encrypted wifi. What I found to be working is:
     1. Move back the original /lib/firmware/iwlwifi-5000-5.ucode into /lib/firmware
     2. do not remove the ko files in the **update** folder
     3. reboot

Then I can connect to internet after it reboots...quite strange.

dpward, 2015-08-07:
There's one missing step: when you add or remove the driver files (*.ko) in the updates directory, you need to run `sudo depmod` afterwards. Does that make a difference?

linkin8834, 2015-08-07:
Hi David,
    Yes I have depmod, after I remove the *ko files from the updates directory. I found that, after step 1 to step 6 as I mentioned above, when I reboot, I can get back to Internet.

linkin8834, 2015-08-07:
Here is the dmesg output:
```
$ dmesg | grep iwlwifi
[   11.809565] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[   11.809615] iwlwifi 0000:03:00.0: irq 52 for MSI/MSI-X
[   11.837604] iwlwifi 0000:03:00.0: loaded firmware version 8.83.5.1 build 33692 op_mode iwldvm
[   11.892573] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[   11.892576] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[   11.892577] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[   11.892579] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[   11.892650] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[   12.641752] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[   12.644783] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[   12.780347] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[   12.783413] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  148.603919] iwlwifi: module verification failed: signature and/or  required key missing - tainting kernel
[  148.604601] iwlwifi: module has bad taint, not creating trace events
[  148.604761] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  148.604821] iwlwifi 0000:03:00.0: irq 51 for MSI/MSI-X
[  148.605064] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[  148.605066] iwlwifi 0000:03:00.0: Falling back to user helper
[  148.605362] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[  148.605364] iwlwifi 0000:03:00.0: Falling back to user helper
[  148.605709] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[  148.605711] iwlwifi 0000:03:00.0: Falling back to user helper
[  148.608376] iwlwifi 0000:03:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[  148.624896] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[  148.624901] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[  148.624902] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[  148.624904] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[  148.625008] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[  156.147559] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  156.147616] iwlwifi 0000:03:00.0: irq 51 for MSI/MSI-X
[  156.147950] iwlwifi 0000:03:00.0: loaded firmware version 8.83.5.1 build 33692 op_mode iwldvm
[  156.157404] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[  156.157407] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[  156.157409] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[  156.157411] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[  156.157462] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[  156.426282] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[  156.429479] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  156.564312] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[  156.567373] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
```

linkin8834, 2015-08-07:
Also, I find that by typing
1. modprobe -r iwlwifi
2. modprobe iwlwifi

I can no longer connect to encrypted WiFi access point, and needs to reboot...

dpward, 2015-08-07:
Ok thanks. It looks like the original driver and firmware are being reloaded fine. I think it may have to do with NetworkManager instead.

Are you using NetworkManager or are you disabling it when using the modified driver/firmware?

dpward, 2015-08-07:
Hmmm... when you type those two commands, and it can't connect to an encrypted AP anymore, is that happening even if you don't ever load the modified driver or firmware after the system is booted?

linkin8834, 2015-08-07:
Hi David, 
     When I am using modified driver/firmware, I disable the network manager, and putting these lines into /etc/network/interfaces:
 ```
auto lo
iface lo inet loopback
iface wlan0 inet manual
```

When I want to use WiFi, I put these lines into /etc/network/interfaces/:
```
auto lo
iface lo inet loopback
```

dpward, 2015-08-07:
Ok, and you are restarting NetworkManager after you update that file, correct?
Is the issue that you see available APs with NetworkManager, but you can no longer connect to them?

linkin8834, 2015-08-07:
Hi David,
    Yes, even if I use the original firmware and driver, it would not connect to encrypted wifi anymore after modprobe -r iwlwifi and modprobe iwlwifi.

linkin8834, 2015-08-07:
Hi David,
     I restart the network manager. I can see APs with networkmanager, but I cannot connect to them. I kept getting *wlan0: deauthenticating from xxxx (reason 3)* in the dmesg.

linkin8834, 2015-08-07:
I think I should try a manual connection instead of using network manager.

dpward, 2015-08-07:
Ok, I'm not sure why this isn't working, all of your steps seem to be correct. Let me know what happens with a manual connection.

linkin8834, 2015-08-09:
HI David,
     Manual connection to AP after modprobe works very well. The catch is to use wpa_supplicant instead of network-manager for WPA encrypted APs.

closed 2015-08-09 by linkin8834