dhalperi/linux-80211n-csitool-supplementary #93: Injection with standard iwlwifi driver and firmware?

linkin8834, 2015-08-15:
Hello,
       I think that LORCON works with standard iwlwifi driver and firmware, since Intel 5300 supports monitor mode. As far as I understand the modified driver and firmware, the only drawback if I use standard firmware is that I cannot set the HT data rate specified in 802.11n for the injector. 
      In the old LORCON code, it seems that they provide support for MIMO and MIMO greenfield for 802.11n (Strangely enough, they don't support 802.11n data rate in the tx80211.h file. All rates are 802.11a/g ). when I tried to get the capabilities in LORCON, it reported that
       [1] I cannot set the TX data rate
       [2] I cannot set the modulation to MIMO or MIMOGF
      With wireshark, I observe that my injection packages are sent with 6Mbps rate, with 'HT information' =0 (so obviously, not 802.11n packages). With modified firmware, I can easily get to 72Mbps with HT information bit = 1. 
      So, my question is, could it be possible that I inject some raw data frames with 802.11n enabled, using the driver iwlwifi, or mac80211, or nl80211? I saw some posts using iwlagn + HT but iwlagn is removed from the newer kernel version of Ubuntu.
     Thanks!

dpward, 2015-08-16:
There's not special code in LORCON to handle the iwlwifi driver, so you are using the mac80211 framework for controlling the adapter. The old version of LORCON does not support setting the rate when using mac80211&mdash; it only does this with special support for the adapter, such as for AirPcap.

You could possibly look at [LORCONv2](https://code.google.com/p/lorcon/source/browse/), but I can't tell easily if it supports setting the rate for mac80211 framework drivers, or if the iwlwifi driver would support that even if it did.

linkin8834, 2015-08-16:
Hi David,
    Thanks for your suggestion. So if I could manually generate a 802.11 radiotap header with 'HT information' bit = 1 and provide correct MCS information (to force 802.11n transmission), it could still be rejected by the iwlwifi driver? 
    Actually, what I found from wireshark when I use the modified firmware is that: in the radiotap header, the *rate flag* is **disabled** in the header, but the *HT flag* is **enabled**, and so there is a three bytes MCS field, specifying the MCS index in 802.11n and other info. Things become quite different when I switch back to standard driver and firmware, where the *rate flag* is **enabled**, and the *HT flag* is **disabled**. So, there is no MCS field and thus, not a 802.11n package.
   For the LORCON-old, they said in the header file that the MIMO capability would be supported in the future. I would definitely try LORCONv2 and see what happens.
   Thanks again!

dpward, 2015-08-16:
The iwlwifi driver gives the firmware the command to transmit a frame, specifying the rate in the format that the firmware understands. The firmware doesn't use the radiotap header. You're limited by the ways in which the driver decides how it will specify the rate to the firmware when transmitting a frame.

linkin8834, 2015-08-16:
Dear David,
     I see. Thanks so much :D

closed 2015-08-16 by linkin8834