dhalperi/linux-80211n-csitool-supplementary #94: rate format learning

hmlsi, 2015-08-16:
Hi,
Can someone provide me reading material from where I can learn about what is format of setting rate. e.g. How to describe this format 0x1c90e? I have read FAQ#1 and related issues too. From https://en.wikipedia.org/wiki/IEEE_802.11n-2009#Data_rates I am unable to understand how to know about which part of rate represents MCS, spatial stream, modulation type etc.

Thanks

dpward, 2015-08-16:
The format of the "rate" value is specific to the iwlwifi driver in the Linux kernel &mdash; you won't find it anywhere else. There's a link in FAQ 1 to the part of the driver source code that describes this format with plenty of comments: https://github.com/dhalperi/linux-80211n-csitool/blob/csitool-3.13/drivers/net/wireless/iwlwifi/dvm/commands.h#L245-L334

The slightly tricky part is that the format of the lower bits is different depending on if the rate is high-throughput (802.11n) or legacy (802.11a/b/g). So it's easier to deconstruct the rate starting with the higher bits.

Let's take your example, `0x1c90e`.

Here are lines 329-331 in the source code I linked to above:
```
#define RATE_MCS_ANT_A_MSK	0x04000
#define RATE_MCS_ANT_B_MSK	0x08000
#define RATE_MCS_ANT_C_MSK	0x10000
```
These values are bit masks that indicate if antenna A, B, or C is used, respectively. If all three antennas are used, you would put them together and get `0x04000 | 0x08000 | 0x10000 = 0x1c000`. This matches the first part of your example rate.

Now let's work backward through the source code. Here are lines 307-309:
```
/* Bit 11: (1) Use 40Mhz HT40 chnl width, (0) use 20 MHz legacy chnl width */
#define RATE_MCS_HT40_POS 11
#define RATE_MCS_HT40_MSK 0x800
```
and lines 294-297:
```
/* Bit 8: (1) HT format, (0) legacy format in bits 7:0 */
#define RATE_MCS_FLAGS_POS 8
#define RATE_MCS_HT_POS 8
#define RATE_MCS_HT_MSK 0x100
```
If I put these two values together, I have `0x800 | 0x100 = 0x900`, which matches the next part of your example rate. So, you have a high-throughput (802.11n) rate that uses a channel width of 40MHz. Note also the other bits that were *not* set, such as bit 13: this means you have a normal guard interval (800 ns).

Since this is a high-throughput rate, we finally look at lines 254-268:
```
 * High-throughput (HT) rate format for bits 7:0 (bit 8 must be "1"):
 *  2-0:  0)   6 Mbps
 *        1)  12 Mbps
 *        2)  18 Mbps
 *        3)  24 Mbps
 *        4)  36 Mbps
 *        5)  48 Mbps
 *        6)  54 Mbps
 *        7)  60 Mbps
 *
 *  4-3:  0)  Single stream (SISO)
 *        1)  Dual stream (MIMO)
 *        2)  Triple stream (MIMO)
 *
 *    5:  Value of 0x20 in bits 7:0 indicates 6 Mbps HT40 duplicate data
```

This part is actually just the MCS index in 802.11n &mdash; which are listed in the table that you linked to . Since `0x0e = 14` you are using MCS 14. Since you have a 40 MHz channel and 800 ns guard interval, as found earlier, your data rate is 243 Mbps.

Note, MCS 14 is dual-stream rate, although you are using all three antennas. The opposite is not possible &mdash; you can't have a triple-stream rate with only two antennas. Be careful not to set an invalid rate such as that, as it would very likely crash the driver.

hmlsi, 2015-08-17:
Thank you so much for such detailed answer. Much appreciated. According to this logic, 0x4101 can be explained as
0x4101=
0x4000      ( RATE_MCS_ANT_A_MSK)
|0x0100     (RATE_MCS_HT_MSK) bit 8 and bit 9 are 0
|0x0001     ( 48 Mbps) (if I refer to 270-278)

Am I getting your point right?

dpward, 2015-08-17:
Your explanation is correct, but you are looking at the wrong column on the Wikipedia page &mdash; look under 20 MHz channel instead of 40 MHz channel. So the rate is 6.5 Mbps (not 13 Mbps).

hmlsi, 2015-08-17:
6.5 Mbps corresponds to MCS index 0 and for MCS index 1, its 13 Mbps. Isn't it?

dpward, 2015-08-17:
Oh sorry, you're right.

closed 2015-08-17 by hmlsi

hmlsi, 2015-09-08:
Hi Sir David,
I want to confirm one thing. for 0x420A, rate will be 1Mbps because its bit 9 is 1 so we'll follow lines 280-284 of code. Am i right?

dpward, 2015-10-02:
Yes, exactly.