dhalperi/linux-80211n-csitool-supplementary #96: Signal Processing

muneebr1, 2015-08-19:
Hi, I have been successful in setting injection mode and receiving csi packets. I have a confusion. Tha packets, which I am getting, are in frequency domain. But this is not making sense to me since the packets are with respect to sub carrier index. What is the actual frequency of sub carriers? I am using a 20 MHz channel. Am i right if i say the sampling frequency is 20 MHz too? Suppose if I want to take ifft of the received packets in Matlab, how would I proceed?

P.S. I know that 20 MHz channel has 64 sub carriers and each sub carrier is 312.5 KHz apart

dpward, 2015-08-20:
Please see [FAQ 6](http://dhalperi.github.io/linux-80211n-csitool/faq.html) and please read the text surrounding the table that it references.

The signal processing is done entirely inside the adapter; all that is exposed by the adapter are the CSI matrices.

linkin8834, 2015-08-20:
You could interpolate you freq domain data since only 30/64 are reported for 20MHz. Then you should do a **fftshift** in MATLAB. Then, just do a **ifft** to convert the data into time domain.

The actual frequency are given as center_freq+subcarrier_index*312.5k, where subcarrier_index is in the range of [-32,31].

lixiaohai, 2015-10-15:
@linkin8834

since you said we can interpolate our freq domain data,could you tell me some details?
what's more,if I do this way whether the results of the IFFT is accurate or not？

Thanks,

linkin8834, 2015-10-15:
@lixiaohai  You could get 30 CSI on 30 subcarriers, defined by 802.11n. The subcarrier index of the 30 subcarriers are defined therein. You could invoke the interp1 function in MATLAB to get the remaining 34 subcarriers (since in total, there are 64). 

For the accuracy, I think the 30 subcarriers out of 64 on 20MHz is okay for IFFT transform. For 40MHz, it is a little bit too short as you are missing 3 subcarriers for each of the 4 subcarriers.

 If your goal is to get a highly accurate picture of time-domain channel, I suggest USRPs with capability of at least 20MHz baseband frequency (such as N210). In USRP, you can define the usable subcarriers in any manner you like. You could search for gr-ieee802-11. It is a Github project as well. You need to write a simple channel estimator though.

lixiaohai, 2015-10-15:
@linkin8834

well,I got it!Much appreciation for your kind reply!!

linkin8834, 2015-10-15:
@lixiaohai  In USRP, you could also simply sends an impulse with a high-enough sampling rate and get the channel response at the receiver. It might be unnecessary to use the OFDM channel estimator. Anyway, you can try the interpolation method for 5300 20MHz and 40MHz and see the performance.

lixiaohai, 2015-10-15:
@linkin8834

sorry,I have another problem.where can I find the defined subcarrier index.I have readed #55,but the link https://standards.ieee.org/getieee802/download/802.11-2012.pdf seems invalid.

linkin8834, 2015-10-15:
@lixiaohai 
BW=20:
usable_sc=[-28:2:-2,-1,1,3:2:27,28];
Nfft=64;

BW=40:
usable_sc=[-58:4:-2,2:4:58];
Nfft=128;

lixiaohai, 2015-10-15:
@linkin8834

Thanks a lot for your patience and timely response

lixiaohai, 2015-10-15:
@linkin8834

I followed as you said,and figure it out.Here is my result of interpolation.Could you tell me whether it is right or not?
[image]

lixiaohai, 2015-10-15:
@linkin8834

sorry,I made a mistake when I deal with the Indexs.I will try again.

lixiaohai, 2015-10-15:
@linkin8834

I correct the error indexs but the result is similar as the picture showed.I want to know why the value of the last subcarrier is so large?Is it normal?To the best of my knowledge,only when the freq domain data is Complex conjugate symmetric can the result of IFFT be real.Is there anything wrong or the date I collect is invalid?

lixiaohai, 2015-10-15:
@linkin8834

Excuse me.I searched a matlab fourm and it said when I invoke the interp1 function the index must be in the range of the X.If so,how can I get the values out of the range(such as index -30)?Can I only use the indexs from -28 to 28 for IFFT?

dhalperi, 2015-11-21:
Sounds like this has turned into a question about how to use MATLAB to do certain types of interpolation given uneven samples. I don't think this is a question that I can answer.

If you do have a useful script that does this type of processing, please submit a pull request!

closed 2015-11-21 by dhalperi