dhalperi/linux-80211n-csitool-supplementary #97: Similar phase response on different locations in master mode

linkin8834, 2015-08-20:
Hello,
       I configure one 5300 laptop as AP using hostapd. The client is another 5300 laptop. Both laptops are running the modified driver and firmware. I did not set the rate for the connected station. AP logs the bfee to a file. I could get a lot of 1 TX stream data at the AP side. However, when I take a look at these data using MATLAB, I saw almost the same phase response, even I move the AP to different locations.
       I have attached a figure here for one link of my measurement, where different colors stands for different locations. As you can see, even I move the AP to different locations, the phase on that link remains identical.
     Thanks!

dpward, 2015-08-21:
Regarding ad-hoc mode, this patch to the iwlwifi driver was necessary for it to work:
https://patchwork.kernel.org/patch/5285661/

I haven't addressed any of the comments yet for it to be included in the upstream kernel. However in my own testing for several weeks I did not experience issues using this.

To apply this patch so that you can use ad-hoc mode, add one command to the installation instructions in step 2, right before "Build the modified driver for the existing kernel":
```
wget -O - https://patchwork.kernel.org/patch/5285661/raw/ | patch --no-backup-if-mismatch -p1
```

linkin8834, 2015-08-21:
Hi David,
    Thanks for the patch. I will apply it to the iwlwifi driver.

linkin8834, 2015-08-21:
Hi David,
     I am on the commit dda0665 in the folder linux-80211n-csitool. I applied your command. It returned as '1 out 2 hunks FAILED --Saving rejects to file drivers/net/wireless/iwlwifi/dvm/mac80211c.rej'.

linkin8834, 2015-08-21:
Hmm, it seems that mac80211.c was patched successfully. I have a new line of sta_rc_update=iwlagn_mac_sta_rc_update. Is the hunks failed thing normal behaviour? Thanks!

dpward, 2015-08-21:
No, that shouldn't be failing...if I check out that same commit and apply the patch, the output I get is:
```
patching file drivers/net/wireless/iwlwifi/dvm/mac80211.c
Hunk #1 succeeded at 940 with fuzz 2.
Hunk #2 succeeded at 1622 (offset -4 lines).
```

Could you run
```
git reset --hard csitool-3.16
git clean -d -x -f
```
and try applying the patch again?

linkin8834, 2015-08-21:
Works. Thanks David!

linkin8834, 2015-08-31:
Hi David,
I patch both laptops with your adhoc patch. I turned on the modified driver with modprobe. Then, I set up the two laptops into IBSS mode with
```
Terminal1:

iw wlan0 del
iw phy phy0 interface add wlan0 type ibss
ifconfig wlan0 10.1.1.1 netmask 255.255.255.0
iw dev wlan0 ibss join tmp 2412 HT20 00:11:11:11:11:11

Terminal2:

iw wlan0 del
iw phy phy0 interface add wlan0 type ibss
ifconfig wlan0 10.1.1.2 netmask 255.255.255.0
iw dev wlan0 ibss join tmp 2412 HT20 00:11:11:11:11:11
```

Then, at laptop A, I type
```
echo 0x1c110 | tee rate_scale_table
```
with root permission. Then, using laptop A, I ping laptop B with
```
ping -i 0.01 10.1.1.2
```
while in another terminal, I run **log_to_file** binary. However, I saw none of the bfee packages a recorded. 

Should I use iperf instead of ping in adhoc mode?

Thanks!

linkin8834, 2015-08-31:
Hi David,
   I tried iperf3 and still, cannot log anything in adhoc modes :(

linkin8834, 2015-09-04:
Hmm, when I applied the patch, the csi tool seems to not working. The packet injection is working, and the monitor laptop could see the packages in wireshark, but failed to record any CSIs. After restoring the old kernel code and build the .ko files again, the log_to_file works normally.

dpward, 2015-09-26:
Sorry for the long delay, I will try to look at this soon.

dhalperi, 2015-11-21:
Closing old issue as it's been 2 months. Has this problem been resolved?

closed 2015-11-21 by dhalperi

linkin8834, 2015-11-21:
@dhalperi  I decide to give up the experiments of using other non-5300 devices for CSI extraction. The link is not stable and the device might use rate adaption. I still prefer the inj/mon mode :D

dhalperi, 2015-11-21:
That follows with my experience. Good to hear, thanks!

swashah, 2016-11-07:
@dpward @linkin8834 @dhalperi 
I tried the patch but couldnt log csi. Any suggestions?

swashah, 2016-11-14:
@linkin8834 : were you able to figure out the issue of CSI amplitude? I am currently trying to collection CSI on 02 laptops very close to one another but receiving different CSI amplitude. Do you have any idea? I am using monitor mode on 02 laptops and third one is injecting packets- Both the laptops (receiving ) are using same single antenna and transmitter is also using a single antenna so its is a SISO system with rate controlled from transmitter side as well