dhalperi/linux-80211n-csitool-supplementary #98: where is the code that calculates the original CSI matrix in the project?

lidongmeng, 2015-08-21:
@dhalperi When I read the paper"Tool Release: Gathering 802.11n Traces with
Channel State Information", I found in the part of "HOW IT WORKS" that "Our firmware modifications enable an Intel debug mode that records CSI for each correctly received 802.11n packet and sends it up to the kernel driver on the host computer. The driver in turn passes the CSI to a user-space program for processing". As it says, your firmware modifications enable the Intel debug mode that records CSI, but I can not  found the code in the csi-tool(as u see, the code is too much), so can u tell me which file should I read in the project. Thank u very much!

dpward, 2015-08-21:
The source code to the firmware is not available; see [FAQ 9](http://dhalperi.github.io/linux-80211n-csitool/faq.html).

lidongmeng, 2015-08-23:
@dpward Oh, it is really a pity! I think when u modify the firmware, u must read something about CSI,( for example, the standard specification about CSI). As I am not familiar with the region, I can not find the paper about how CSI is specified, so could u give me some suggestion or paper name about CSI?Thank u very much!

dpward, 2015-08-23:
See the links on the [home page](http://dhalperi.github.io/linux-80211n-csitool/): you should certainly read the [tool release announcement](http://www.halper.in/pubs/halperin_csitool.pdf) and you may also want to read the [SIGCOMM 2010 paper](http://www.halper.in/pubs/comm356s-halperin.pdf).

You can also refer to the [IEEE 802.11-2012 standard](http://standards.ieee.org/getieee802/download/802.11-2012.pdf) itself (this includes the 802.11n amendment and is freely available). Section 20.3.12.3 describes how the CSI matrix *H<sub>eff</sub>* is measured, and Table 8-45 lists the subcarriers that are measured.

lidongmeng, 2015-08-24:
@dpward Thank u very much!

dhalperi, 2015-08-24:
This computation is not in the firmware, it's in *hardware*. The firmware is basically a hardware "driver" more than anything else.

@dpward I feel like we (you, these days) spend a lot of time re-hashing FAQ entries. Any suggestions? Also feel free to update them.

closed 2015-08-24 by dhalperi

lidongmeng, 2015-08-27:
@dhalperi In other word, the hardward receives the packet from the TX, and then decodes the packet  and computes the CSI values, then the firmwire extracts the csi values and sends them to the kernel space, then the userspace? am I right?

dhalperi, 2015-08-30:
The firmware doesn't even extract the CSI values. The firmware just asks the hardware to put the computed CSI values into the buffer that goes to the kernel.