dhalperi/linux-80211n-csitool-supplementary #101: Logging synchronization between two laptops with Intel 5300 cards

cathycy, 2015-09-15:
Hi,
I have two laptops configured with the Intel 5300 cards and csitool, and one of them is in AP mode(configured using hostapd). The client laptop connects to the host laptop. I have both of them logging the CSI values. The client laptop then pings the host laptop, and the pings are answered. The csi is then recorded. 

What I think happens is that, the client laptop sends a ping, the host laptop receives it, and answers that ping, which is then received by the client laptop. The CSI is logged for these receptions at both laptops. This means I should have an equal number of CSI readings at both laptops, and the data should be synchronized except for a small delay in the packet being sent back. 

However, from my observations during experiments, the number of samples logged by the client is very different from the number of samples logged by the host laptop. Also, the samples don't seem to be synchronized. Any suggestions on what I can do to get synchronization between the laptops? Any other suggestions are welcome too!

Thank you!

dpward, 2015-09-26:
In addition to the ICMP (ping) packets that you are sending, you need to consider the control traffic that is sent on the network as well (ARP/Neighbor Discovery, Zeroconf/mDNS, etc.) which can also result in CSI. Also depending on the link conditions, 802.11 retransmissions may occur and you may wind up with samples for each transmission attempt (if an ACK is not received, for example). It's best to examine a packet capture on both sides to make sure you understand the traffic profile (ideally creating another interface in monitor mode and capturing on it with Wireshark, so you can see traffic at the link layer).

cathycy, 2015-10-05:
Thank you for the reply. Since I cannot control the link conditions, is there a better way to do this experiment instead of having AP and client mode on the two laptops? Or do you think I can achieve the required experiment setting only by creating another interface in monitor mode and logging the traffic? I am asking this because I don't have another interface for monitor mode right now and want to know if there is a possibility by using just the current setup.

dhalperi, 2015-10-05:
The way I did this is by putting a sequence number in the payload of the packets, and using that sequence number to synchronize logs.

closed 2015-10-05 by dhalperi

cathycy, 2015-10-07:
Hi, 

Sorry to trouble you again, but I have a few questions since I am new to this:

1. There is a sequence number for the packets, ICMP seq, which is displayed in the terminal when I ping. Can this sequence number be used for the synchronization purpose, or would I have to put in another sequence number in the payload?

2. I read FAQ 12, which says that by enabling IWL_CONN_RX_MPDU_MSK , the IEEE 802.11 sequence number would be logged. I tried reading through the codes to understand where this information would show up, but I am not great at this, and couldn't understand how to extract this information. Could you please explain/ refer me to codes on how to get the sequence number for the packets?

Thank you!

dhalperi, 2015-10-08:
If you set that bit, then the file produced by `log_to_file` will include records with payloads in addition to the other records you are recording. You will then need to modify the processing code to handle them.

If you inspect [this part of `read_bf_file.m`](https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/matlab/read_bf_file.m#L51), you can see that we skip all records other that do not have code `0xbb`, for beamforming measurements.

If you add a handler for the [`0xc1` log entries](https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/netlink/iwl_structs.h#L353), you will see the packet payloads.

Packet payload records are logged immediately following the CSI, which is how you can link CSI entries with individual payloads.

dhalperi, 2015-10-08:
See also David's comment here: https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/65#issuecomment-115325845

@dpward: I thought that CSI preceded MPDU, but you indicate the reverse. I may be wrong – it's been more than 3 years since I ran the CSI tool ;).

swashah, 2016-11-19:
@cathycy : Were you able to synchronize two laptops, as I also want to do synchronization of CSI collected on 02 machines in monitor mode. I will appreciate your help