dhalperi/linux-80211n-csitool-supplementary #103: how to reduce number of packets received in monitor mode?

hmlsi, 2015-10-02:
Hi,
I am using monitor mode to record CSI. As far as I have seen, the minimum rate which I can set is 13Mbps but at this rate I got a lot of packets. if I intend to record data of 10 seconds or more,its like 50k packets. Its very complex to observe them. Any suggestions on how to reduce this large number of packets? Can I set how much packet I want to receive in a fixed time interval?

dpward, 2015-10-02:
Are you using the injection scripts with monitor mode? `random_packets` takes an optional fourth argument, which specifies the delay in microseconds between packets. Use this if you don't want to send packets continuously, which will reduce the number of samples.

hmlsi, 2015-10-05:
Sorry for the delay. Yes, I am using injection scripts with monitor mode

dhalperi, 2015-11-21:
Closing this old issue as obsolete. Summary:

1. In monitor mode, you will overhear packets from everyone transmitting on that frequency. Try to use less-used spectrum. Most universities, for example, ignore large swathes of frequency. (I hope this changes, but it will be sad for researchers when it does ;)

2. Use the delay argument to `random_packets` that @dpward mentioned.

closed 2015-11-21 by dhalperi