dhalperi/linux-80211n-csitool-supplementary #105: when i do the last step---"log_to_file csi.dat"and"ping AP IP ",i can't capture any data.

Darwintam, 2015-10-03:
when i do the last step---"log_to_file csi.dat"and"ping AP IP ",i can't capture any data.

Does the desktop or laptop have an effect on the experiment?

How can i deal with this problem?

thank you !

Tam

dpward, 2015-10-03:
Regarding different system models, the main consideration is that different laptops have a different number/type/placement of internal antennas for Wi-Fi, which will impact the number of simultaneous streams you can achieve, the quality of the link, etc.

In general, your issue is caused either because:

1. the modified driver or modified firmware is not loaded, or
2. your access point is not transmitting packets to you at High Throughput (802.11n) rates.

To troubleshoot 1:
* If the modified driver was not loaded, then this command from step 5 would have failed, because the original driver does not have a connector_log parameter: `sudo modprobe iwlwifi connector_log=0x1`
* Otherwise the modified driver is loaded and we should just check for the modified firmware. Look at the output of `dmesg`; if the modified firmware is loaded, you should see a line that contains the string: `loaded firmware version 8.24.2.12`

To troubleshoot 2:
* Is this a commercial access point (AP) &mdash; and if so, does it support 802.11n (and if applicable, is 802.11n enabled)? Unfortunately the specific transmit rate cannot be controlled on a commercial AP, but typically the rate selection algorithm will switch to a High Throughput rate after initially using a legacy rate if the link quality is good.
* Or are you connecting to another system with this same Intel adapter that is running `hostapd` &mdash; and if so, are you setting the transmit rate on that system to a High Throughput rate (see FAQ 1)?

dhalperi, 2015-11-21:
Closing as obsolete – user never responded.

closed 2015-11-21 by dhalperi