dhalperi/linux-80211n-csitool-supplementary #106: HELP! I find that after"sudo rmmod iwlwifi mac80211 cfg80211 "  ,when i do "sudo modprobe iwlwifi connector_log=0x1 " I can't link WI-FI.

Darwintam, 2015-10-07:
HELP!
 I find that after"sudo rmmod iwlwifi mac80211 cfg80211 "  ,when i do "sudo modprobe iwlwifi connector_log=0x1 " I can't link WI-FI.

How can I solve this difficulty?

THANK YOU!!

Darwintam, 2015-10-07:
root@darwintam-desktop:/proc# dmesg | grep wifi
[   22.164594] iwlwifi: Intel(R) Wireless WiFi Link AGN driver for Linux, in-tree:d
[   22.164598] iwlwifi: Copyright(c) 2003-2012 Intel Corporation
[   22.164653] iwlwifi 0000:01:00.0: pci_resource_len = 0x00002000
[   22.164655] iwlwifi 0000:01:00.0: pci_resource_base = f84ec000
[   22.164657] iwlwifi 0000:01:00.0: HW Revision ID = 0x0
[   22.164709] iwlwifi 0000:01:00.0: irq 45 for MSI/MSI-X
[   22.170999] iwlwifi 0000:01:00.0: Firmware size does not match iwlwifi-5000-2.ucode.sigcomm2010. The UW 802.11n CSI Tool will not work.
[   22.171034] iwlwifi 0000:01:00.0: loaded firmware version 8.24.2.12
[   22.171260] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEBUG enabled
[   22.171261] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[   22.171263] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[   22.171264] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEVICE_TESTMODE disabled
[   22.171266] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_P2P disabled
[   22.171268] iwlwifi 0000:01:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[   22.171316] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[   22.203792] iwlwifi 0000:01:00.0: device EEPROM VER=0x11e, CALIB=0x4
[   22.203795] iwlwifi 0000:01:00.0: Device SKU: 0xF0
[   22.203797] iwlwifi 0000:01:00.0: Valid Tx ant: 0x7, Valid Rx ant: 0x7
[   22.203807] iwlwifi 0000:01:00.0: Tunable channels: 13 802.11bg, 24 802.11a channels
[   22.910505] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[   22.914402] iwlwifi 0000:01:00.0: Radio type=0x0-0x2-0x0
[   23.063616] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[   23.066530] iwlwifi 0000:01:00.0: Radio type=0x0-0x2-0x0
[ 1488.674319] iwlwifi: Intel(R) Wireless WiFi Link AGN driver for Linux, in-tree:d
[ 1488.674327] iwlwifi: Copyright(c) 2003-2012 Intel Corporation
[ 1488.674421] iwlwifi 0000:01:00.0: pci_resource_len = 0x00002000
[ 1488.674423] iwlwifi 0000:01:00.0: pci_resource_base = f96b8000
[ 1488.674425] iwlwifi 0000:01:00.0: HW Revision ID = 0x0
[ 1488.674482] iwlwifi 0000:01:00.0: irq 45 for MSI/MSI-X
[ 1488.683285] iwlwifi 0000:01:00.0: loaded firmware version 8.24.2.12
[ 1488.683526] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEBUG enabled
[ 1488.683528] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[ 1488.683529] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[ 1488.683531] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEVICE_TESTMODE disabled
[ 1488.683532] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_P2P disabled
[ 1488.683534] iwlwifi 0000:01:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[ 1488.683586] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[ 1488.721553] iwlwifi 0000:01:00.0: device EEPROM VER=0x11e, CALIB=0x4
[ 1488.721559] iwlwifi 0000:01:00.0: Device SKU: 0xF0
[ 1488.721563] iwlwifi 0000:01:00.0: Valid Tx ant: 0x7, Valid Rx ant: 0x7
[ 1488.721580] iwlwifi 0000:01:00.0: Tunable channels: 13 802.11bg, 24 802.11a channels
[ 1488.730314] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[ 1488.735247] iwlwifi 0000:01:00.0: Radio type=0x0-0x2-0x0
[ 1488.881534] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[ 1488.884467] iwlwifi 0000:01:00.0: Radio type=0x0-0x2-0x0
[ 3306.288542] iwlwifi: Intel(R) Wireless WiFi Link AGN driver for Linux, in-tree:d
[ 3306.288547] iwlwifi: Copyright(c) 2003-2012 Intel Corporation
[ 3306.288629] iwlwifi 0000:01:00.0: pci_resource_len = 0x00002000
[ 3306.288633] iwlwifi 0000:01:00.0: pci_resource_base = fb938000
[ 3306.288636] iwlwifi 0000:01:00.0: HW Revision ID = 0x0
[ 3306.288699] iwlwifi 0000:01:00.0: irq 45 for MSI/MSI-X
[ 3306.298764] iwlwifi 0000:01:00.0: loaded firmware version 8.24.2.12
[ 3306.299090] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEBUG enabled
[ 3306.299093] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[ 3306.299094] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[ 3306.299096] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_DEVICE_TESTMODE disabled
[ 3306.299097] iwlwifi 0000:01:00.0: CONFIG_IWLWIFI_P2P disabled
[ 3306.299099] iwlwifi 0000:01:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[ 3306.299152] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[ 3306.332238] iwlwifi 0000:01:00.0: device EEPROM VER=0x11e, CALIB=0x4
[ 3306.332242] iwlwifi 0000:01:00.0: Device SKU: 0xF0
[ 3306.332244] iwlwifi 0000:01:00.0: Valid Tx ant: 0x7, Valid Rx ant: 0x7
[ 3306.332255] iwlwifi 0000:01:00.0: Tunable channels: 13 802.11bg, 24 802.11a channels
[ 3306.343857] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[ 3306.347970] iwlwifi 0000:01:00.0: Radio type=0x0-0x2-0x0
[ 3306.494531] iwlwifi 0000:01:00.0: L1 Disabled; Enabling L0S
[ 3306.497461] iwlwifi 0000:01:00.0: Radio type=0x0-0x2-0x0

dpward, 2015-10-07:
```
[ 22.170999] iwlwifi 0000:01:00.0: Firmware size does not match iwlwifi-5000-2.ucode.sigcomm2010. The UW 802.11n CSI Tool will not work.
```

1. You have loaded the modified driver, but it's not able to load the modified firmware. Can you make sure that you relocated the original firmware as described in step 3 of the [installation instructions](http://dhalperi.github.io/linux-80211n-csitool/installation.html)? Specifically, there should no longer be a file named `/lib/firmware/iwlwifi-5000-5.ucode`; it should have been renamed.

2. Is your issue that you can't connect to a normal Wi-Fi access point anymore (which happens to use WEP/WPA/WPA2 encryption)? See [FAQ 7](http://dhalperi.github.io/linux-80211n-csitool/faq.html).

Darwintam, 2015-10-08:
I got it ! thank you very very very much !  If you come to China I invite you to  dinner    Ahahahah!   However ,i still have a problem , without wl5300  ,If I use the other card, can i collect  the CSI data ?

dhalperi, 2015-10-11:
The only card that this tool works with is the IWL 5300 – See FAQ # 4.

closed 2015-10-11 by dhalperi

tfsanam, 2017-04-07:
Darwintam:
 Same as you, when i do "sudo modprobe iwlwifi connector_log=0x1 " I can't link WI-FI.

So what was wrong in your case? How did you fix that?

Thanks