dhalperi/linux-80211n-csitool-supplementary #107: Kindly help me

anujsakhardande14, 2015-10-10:
I am new to this site and I need help in using the dhalperi/linux-80211n-csitool-supplementary.
I am a Masters student from Nanyang Technological University, Singapore.
I am doing dissertation on CSI based key generation technique. I would like to use this tool for the same. 
Kindly help me in understanding the tool, its installation and operation of the tool. Thank you
Please revert.

Thank you ,
Anuj Sakhardande

ozawali, 2015-10-11:
you can read the FAQ

closed 2015-10-11 by dhalperi

tahmidzbr, 2016-10-05:
@anujsakhardande14  Hi im a student at UBC in Canada and working for a project in Lily which is in your uni, NTU. I am also using wifi to do some project, do you want to collaborate atleast the data collection part? i.e. how to measure the CSIs? If interested let me know.