dhalperi/linux-80211n-csitool-supplementary #108: address of transmitter  in injection mode

libing64, 2015-10-10:
@dpward  
Hi, David
     I am using injection mode to collection csi data from more than one transmitters. Now I can receive csi data from two transmitters simultaneously, but I don't  know which transmitter the data belongs to.  
Can you give me some advice to get the source address of each csi data?

Thanks.

libing64, 2015-10-11:
At the transmitter part, the packet is generated and sent, but at the receiver part, I can only get csi information, but how to get the bytes in the custom packet?

1. transmitter code to copy custom data to the payload and send it
```
payload_memcpy(packet->payload, packet_size,
				(i*packet_size) % PAYLOAD_SIZE);
ret = tx80211_txpacket(&tx, &tx_packet);
```

2. receiver code to get csi data
```
	int ret = recv(sock_fd, netlink_buffer, sizeof(netlink_buffer), 0);
	if (ret == -1) {
		perror("netlink recv");
		return ret;
	}
	
	/* Pull out the message portion and print some stats */
	struct cn_msg *cmsg = (struct cn_msg*)NLMSG_DATA(netlink_buffer);
/* 	printf("received %d bytes: id: %u val: %u seq: %u clen: %d\n", */
/* 			cmsg->len, cmsg->id.idx, cmsg->id.val, */
/* 			cmsg->seq, cmsg->len); */
	*buf = cmsg->data;
	*len = cmsg->len;
```

Where is the custom data in the payload? could anyone help me?

libing64, 2015-10-11:
@dhalperi @dpward @ippeb

dhalperi, 2015-10-11:
Read the existing issues! https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/101

closed 2015-10-11 by dhalperi

swashah, 2016-11-22:
@libing64 : Were you able to customize packets in injection mode? I want a sequence number with every packet sent and then read. I will appreciate your help.Thanks