dhalperi/linux-80211n-csitool-supplementary #109: about the frequecy of the subcarriers

ozawali, 2015-10-11:
@dpward  I want to know the frequecy of the subcarriers under the 5G with the 40MHZ channel ,though I have searched it for long time I can't find it .can you tell me  or give some materials to slove it Thank you!

dhalperi, 2015-10-11:
Faq #6.

closed 2015-10-11 by dhalperi

asifhanif1122, 2016-05-16:
@dhalperi @dpward @ozawali  @ippeb  According to 802.11 standard for 20MHz bandwidth these sub-carriers are reported (-28,-26,-24,...-4,-2,-1,1,3,5,7,...25,27,28). We know that in 802.11n standard 20MHz bandwidth is divided in 64 equally spaced sub-carriers. But i am having problem in finding exact value of frequencies of sub_carriers. If we select channel 1 (with centre frequency 2.412GHz), then what will be exact value of frequencies for reported sub-carriers in our case. I will be thankful if you guide me or suggest a link showing description about problem concerned.

tahmidzbr, 2016-11-04:
@asifhanif1122  yeah good question, b/c i dont know exactly how is the grouping done? Is it like for example: in 20 MHz 56 subcarriers are actually used (including data and pilot) and grouped in a way to give 30 subcarriers. Is the grouping done like taking the average of 2 subcarrier's amplitude and phase or is it like when grouping them say -28, -26, -24, they take -28 and omit subcarrier -27?
do you know?