dhalperi/linux-80211n-csitool-supplementary #111: phase difference between RX chains is not constant, pls help!

hotdrive, 2015-10-20:
we are doing some AoA localization research on 5300 NIC, when calibrate the initial phase of 3 RX chains, We find : phase difference between RX chains is not constant, it varies like a noise.
our calibration method: use NIC(A) as a TX, connected with a cable and a splitter, the 2 output of splitter connect to 2 RX ports of NIC(B).
I am wondering if the 3 RX chains shares the same clock, the phase difference between RX chains in the same time should be constant?

hotdrive, 2015-10-20:
We also find ,the AGC in the packets is also changing ,will AGC results in the problems?

hotdrive, 2015-10-20:
by the way, the variation is about 10~40 degrees

dhalperi, 2015-10-21:
The [Phaser](https://jon.thesquareplanet.com/papers/mobicom14-phaser.pdf) project found that phase difference between RX chains was only constant at 5 GHz band.

I assume you're looking at CSI output by `get_scaled_csi`, right, not raw? Otherwise you'll get strange receiver antenna selection effects.

And you're hard-coding both the transmit rate and a single transmit antenna, for good measure?

ozawali, 2015-10-21:
@dhalperi  you mean that cannot use the get_scaled_csi ?

hotdrive, 2015-10-21:
@dhalperi,thx for your reply.
1)we already known this problem, so we are using 5GHz band; monitor mode; single transmit antenna
2)I also noticed the antenna selection effects and correct it. It seems the phase difference will not change by using raw CSI or CSI output by  get_scaled_csi .
BTW, what is the right variation of the channel phase difference if power on? 10 degrees? what makes the variation?
Look foward to your reply.

dhalperi, 2015-10-21:
Others have made this work; I suspect your setup has something wrong. You
might ask them, though!

hotdrive, 2015-10-22:
@dhalperi
Will AGC impact the phase? we find that AGC varies in packets.
BTW, which teams have made this work? any paper ? I have read a lot papers, but the answer is the phase difference is constant without any explain.
I also suspect the setup problem.

dhalperi, 2015-10-26:
I already linked a paper: https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/111#issuecomment-149776936

closed 2015-11-21 by dhalperi