dhalperi/linux-80211n-csitool-supplementary #112: about the perm

ozawali, 2015-10-25:
@dpward  @dhalperi  I want to know the data we collect wether have ajust to the order that A B C ?  if not  we must to adjust from the perm of the every packet  right?

libing64, 2015-10-26:
you can read the matlab code in read_bf_file.m, the csi data is  reordered in this line: 

ret{count}.csi(:,perm(1:Nrx),:) = ret{count}.csi(:,1:Nrx,:);

ozawali, 2015-10-26:
@libing64  Thank you but my question is that the get_scale_csi the output data the order of the NRX is that A B C or arrange by the perm

dhalperi, 2015-10-27:
There are two different concepts in the hardware and drivers.

1. Antennas A, B, and C.
2. RX chains A, B, and C.

BEFORE the perm is applied as @libing64 indicates, the CSI is ordered by rx chains A,B,C. The typical behavior for the hardware is NOT, however, to map antenna A to RX chain A and B to RX chain B -- instead, it tends to map them in order of SNR according to perm.

AFTER the perm is applied, the  CSI is ordered by antennas A, B, C. This is what you want for all practical purposes.

dhalperi, 2015-10-27:
You should not have to do anything with the permutations, as long as you follow the recommended path of processing CSI in the FAQ.

closed 2015-10-27 by dhalperi