dhalperi/linux-80211n-csitool-supplementary #113: Relative phase between 2 rx antennas will shift pi or  pi/2

libing64, 2015-10-26:
I am using a NIC card as a transmitter and another NIC as a receiver, and I plot the relative phase between rx antenna 1,2 and between antenna 2,3. 
plot(angle(antenna1 .* conj(antenna2))) 
plot(angle(antenna2 .* conj(antenna3))) 

Sometimes the relative phase will shift pi or pi/2.
@dhalperi @dpward @ippeb @xu-wang11
can anyone give me some advice to figure it out or solve this problem?

[image]
[image]
[image]
[image]
[image]

[image]

ozawali, 2015-10-26:
you do this job on the 2.4G or 5G?

ozawali, 2015-10-26:
on the 5G I haven't this problem and it is the constant

libing64, 2015-10-26:
This is the configuration command：    
transmitter configuration:
        ./setup_inject.sh 64 HT20
         echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate`
         sudo ./random_packets 100000 100 1 1000
 
     receiver configuration:
         ./setup_monitor_csi.sh 64 HT20

I think the frequency of channel 64 is 5.32GHz.
Is the configuration right? @ozawali

ozawali, 2015-10-26:
how to receive ? splitter?

libing64, 2015-10-26:
I receive the package with log_to_file, and plot it with in ros.
What you mean by splitter? Sorry I don't get the point. @ozawali

ozawali, 2015-10-26:
你所看到的相位差是你直接用空中传输得到的吗？ 如果是那样相位差不可能恒定

libing64, 2015-10-26:
The phase is not constant but the relative phase between 2 rx antennas should be constant if they are kept static.  @ozawali

libing64, 2015-10-26:
I think I have solved this problem.
According to issue [#6] (https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/6) 
On 2.4 GHz bands, it appears that each antenna experiences an unknown phase shift by a multiple of π/2 (or 90 degrees). 

Through I have configured the channel, it seems doesn't work.
When I verify the configuration with 
      #iwconfg
The frequency is still 2.4G.
After configure the channel with iwconfig
     #iwconfig wlan0 channel 64
the frequency becomes 5G, then the shift disappears.

libing64, 2015-10-26:
Thanks @ozawali  and @xu-wang11 for your kindly help.

libing64, 2015-10-27:
@dhalperi @dpward , Now you can close the issue, thanks.

closed 2015-10-27 by dhalperi

shaozhuchuang, 2016-03-11:
@libing64 
#iwconfig wlan0 channel 64 
Can you tell me this command is executed in recevier or transimitter?

libing64, 2016-03-11:
@shaozhuchuang  the both.

shaozhuchuang, 2016-03-11:
@libing64 I also do an experiment and collect some csi information, but i found the data is very not stable, I don't know whether my data and my matlab code have some problems, Can you send your csi files and relate matlab code to me, My email is shaozhuchuang@126.com or mrshao@sjtu.edu.cn. if you are inconvenient to send matlab code to me, i hope you can only send csi fils. Thank you!

CrystalWong2, 2016-09-14:
@libing64 @ozawali Sorry to bother you both. I try to  plot the relative phase between rx antenna 1,2 and between antenna 2,3. And the result is constant which is 0. Am I right？ I use AP as transmitter and NIC as receiver. And I work in 5G 20MHZ.