dhalperi/linux-80211n-csitool-supplementary #114: about the csi_trace

haimapi, 2015-11-04:
hello, I am confused that the csi_trace contains just 73 structures, or less. Is this a commom problem or because of my setting? Thanks for help!

[image]
[image]

dhalperi, 2015-11-04:
Each entry in csi_trace is a single CSI measurement. You need to look inside the struct to see the actual values.

Please see FAQ 2: http://dhalperi.github.io/linux-80211n-csitool/faq.html

One relevant excerpt:

```
>> csi_entry = csi_trace{1}	(Note the curly-braces {}, not parentheses ().)

csi_entry =

    timestamp_low: 4            (In the sample trace, timestamp_low is invalid and always 4.)
       bfee_count: 72
              Nrx: 3
              Ntx: 1
           rssi_a: 33
           rssi_b: 37
           rssi_c: 41
            noise: -127
              agc: 38
             perm: [3 2 1]
             rate: 256
              csi: [1x3x30 double]
```

closed 2015-11-04 by dhalperi

chensiwen, 2017-05-11:
@haimapi I have the same case with you. There are always some '[]' in the _csi_trace_. And sometimes, there are above 80% '[]'. Do you know why now?