dhalperi/linux-80211n-csitool-supplementary #115: another question: about the csi_trace

haimapi, 2015-11-04:
Thanks for help, and I've open the struct, e.g the 43rd struct, it's empty.
[image]

dhalperi, 2015-11-04:
Please post the actual interpreter command line and output -- these screenshots are not useful.

Note that `csi_trace{43,1}` is not correct -- it should just be e.g. `csi_trace{43}`.

dhalperi, 2015-11-04:
Also in your second screenshot from #114, `csi_trace{43}` is an empty cell -- only 1 through 42 are valid.

haimapi, 2015-11-04:
I tried to get the CSI:

>> csi_entry=csi_trace{43}

csi_entry =

     []

>> csi=get_scaled_csi(csi_entry)
??? Attempt to reference field of non-structure array.

Error in ==> get_scaled_csi at 7

haimapi, 2015-11-04:
Actuall I make MP ping to the AP for about 1min to get this .dat file. I just feel puzzled that only 42 structures were alivable in a .dat file..

ozawali, 2015-11-04:
you should see the clue the system show you that if you receive the packets.

ozawali, 2015-11-04:
see the all FAQ

dhalperi, 2015-11-21:
If your question is why you aren't receiving packets, please carefully read the [FAQ](http://dhalperi.github.io/linux-80211n-csitool/faq.html), especially number 6.

closed 2015-11-21 by dhalperi