dhalperi/linux-80211n-csitool-supplementary #116: How to set transmit rate on ack packets

alcat88, 2015-11-04:
Hello,
i'm using AP mode (with hostapd) to connect two 5300 devices and my goal is to capture CSI info both on the receiver both on the transmitter(capturing the ack generated by the receiver) using iperf to inject the packets.
I’m able to extract csi info on the receiver because i enabled the HT mode by setting the transmit rate on the transmitter (in this case 0x4102) but i can’t capture anything on the other side because i can’t set the transmit rate for the ack packets generated by the receiver. 
So my question is if it is possible to set the transmit rate/mcs for the ack packets generated by the receiver.
I hope i was clear, 
thanks in advance.

dhalperi, 2015-11-05:
Cannot, I believe.

closed 2015-11-05 by dhalperi

Darwintam, 2015-11-14:
I wanna to improve the sampling rate of catching CSI, do you have any methods?  THANK YOU!