dhalperi/linux-80211n-csitool-supplementary #117: issues with extracting time of flight from CSI phase

manudfsn, 2015-11-06:
I’m trying to get a rough estimate of time of flight by looking at the slope of the CSI phase (unwrapped) across subcarriers. The TOF that I’m estimating doesn’t appear to correspond to the distance between NICs in any way. I’m running in injection mode with a 40 MHz channel. The linear fit to the phase slope indicates a delay on the order of 200 ns. I understand that most of this delay is from packet detection and other delays in Tx and Rx. However, I assume that I should be able to see some time of flight variation on top of the static delay. I’m sending 100 packets and taking the average linear fit for each distance. The standard deviation is typically less than 10 ns.  If I measure the delay for distances between 10 ft and 50 ft,  the difference in delays doesn’t correspond to the difference in distance. The delay vs distance doesn’t actually seem to be monotonic either. I’ve taken the IFFT and looked at the impulse response, and see similar results to the simple linear fit. Any idea why the TOF estimated from CSI doesn’t make sense?

hotdrive, 2015-11-10:
as you mentioned above, packet detection causes this result and the slope just means the group time delay, not the actual ToF.i think there is no way to estimate the exact TOF

dhalperi, 2015-11-21:
Sorry, not my error of expertise :)

closed 2015-11-21 by dhalperi