dhalperi/linux-80211n-csitool-supplementary #118: can't capture CSI data

star90, 2015-11-07:
Hi, guys,

I am writing to ask help for capturing CSI data. 

Basically, i am using lenovo X200 Tablet with a Intel 5300 card, and the operating system is Ubuntu  11.04. I have done all the steps in the new installation instructions, but I still can't capture CSI data. I can produce the file, like tmp.dat, but it is empty (0 bytes).

I have looked through all the issues, and I have tested following things:
1. modified driver.  
   The command 'sudo modprobe iwlwifi connector_log=0x1' works, so the modified driver is fine.
2. modified firmware. 
   After running 'dmesg', I can see 'loaded firmware version 8.24.2.12'. So the modified firmware is fine.
3. the mode of AP
    I have tried the 'bgn mixed' mode and also 'n only' mode, but still have only have empty file. 

I don't know what is the problem, and I am eagerly looking for help. 

Best wishes for all!!

dhalperi, 2015-11-13:
You did not describe the data that you're sending, nor how long you sent data for.

Did you try using something like Wireshark to inspect the rates of the packets received? Make sure you see packets received at 802.11n rates that meet all requirements laid out in the FAQ for CSI measurement.

closed 2015-11-21 by dhalperi

tahmidzbr, 2016-10-19:
@star90 hi i can see the packets i receive are recieved at 802.11n rate. But still i cant see anything being logged to log_to_file csi.dat. Can someone please explain?

FTB-B, 2016-11-22:
@tahmidzbr Hi, could you solve your problem? I have the same problem, my csi.dat is also empty. Can anybody help me please.

tfsanam, 2017-04-07:
@tahmidzbr how can you check/see the packets you receive are recieved at 802.11n rate?
@dhalperi How can we make  sure thatI see packets received at 802.11n rates? what command/procedure?

ljk8918, 2018-06-14:
@tahmidzbr  @FTB-B  @tfsanam @dhalperi 
I have the same problem. My csi.dat is empty. anyone solve this problem??