dhalperi/linux-80211n-csitool-supplementary #119: if I want to improve the sampling rate , what should i do ??

Darwintam, 2015-11-14:
if I want to improve the sampling rate , what should i do ??    FOR EXAMPLE  ,now i catch the csi by ping <AP IP>,which is about 1 sample/second. However , i want to catch 800 sample/second now. what should i do?  

THANK YOU!

ozawali, 2015-11-14:
you should use the injection mode

dhalperi, 2015-11-15:
I agree with @ozawali – the best thing to do for rapid sampling is to use injection mode with short packet payloads. See the instructions here: https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection

closed 2015-11-15 by dhalperi

haimapi, 2015-12-10:
will it work if i set the ping interval at 0.1 second?
eg, i type: ping -i 0.1 192.168.1.1 in the terminal?

dhalperi, 2015-12-10:
That will definitely increase the rate at which packets are transmitted.

However, I have generally found it makes life much, much, much easier to do all my research experiments in injection mode.