dhalperi/linux-80211n-csitool-supplementary #120: About csi_trace

haimapi, 2015-11-18:
Dear all,
        Yesterday, when I process CSI data with matlab on another laptop, a question occur:
[image]

I've read FAQ and recompile read_bfee.c with the command:

ozawali, 2015-11-18:
you should recomplie see FAQ 2A

haimapi, 2015-11-18:
@ozawali , I just want to know how to recompile readf_bfee.c.... Isn't it "mex read_bfee.c"?

ozawali, 2015-11-18:
right !

haimapi, 2015-11-18:
@ozawali , yeah, I did as told, after typing "mex read_bfee.c" it threw out errors and recompile failed...

dhalperi, 2015-11-21:
What version of MATLAB are you using?

I've only compiled the file using a specific, fairly new, MATLAB on OS X. There may be platform-dependent changes you need to make to the source `read_bfee.c` file.

dpward, 2015-11-23:
@haimapi @dhalperi The error is caused because your C compiler (presumably Microsoft Visual C++) does not support non-constant initializers for arrays, which was [introduced in C99](http://www.drdobbs.com/the-new-c-declarations-initializations/184401377).

Microsoft Visual C++ has long been lacking in support for C99 (at least [until very recently](http://blogs.msdn.com/b/vcblog/archive/2013/07/19/c99-library-support-in-visual-studio-2013.aspx)). That being said, I'm actually not sure why I didn't see this error before when I last built the MEX-file for Windows (is this an older version of MATLAB / Windows / Microsoft Visual C++?).

I will see if I can modify that code soon so it is C89 compliant and will work for you (I've been very inactive on this lately).

haimapi, 2015-12-01:
@dhalperi @dpward thx very much for your help! the matlab version i use is '2010b'

asifhanif1122, 2016-01-26:
I have noticed that code does not work on Matlab versions except for 2010 version. If anyone is experiencing this "??? Undefined command/function 'read_bfee' " error, he/she should use Matlab 2010 version.

egaebel, 2016-02-26:
So it isn't possible to compile on any version except 2010a??? There must be some workaround...

asifhanif1122, 2016-02-29:
I experienced these problems while executing files on 64 bit version of Matlab. It is possible that files may work on other versions of Matlab (32-Bit).

egaebel, 2016-02-29:
Update: I was able to successfully compile on Matlab 2014a, 64-bit. So I don't think this is a general problem

zainalabidenakhtar, 2016-05-21:
i am also facing the same problem but not got any solution yet. if any body got this problem and then resolved he may kindly guide

zainalabidenakhtar, 2016-05-21:
>> csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6');
??? Undefined function or method 'read_bfee' for input arguments of type 'uint8'.

Error in ==> read_bf_file at 66
        ret{count} = read_bfee(bytes);
 
>> mex read_bfee.c
 
Select a compiler: 
 
[0] None 
 
Compiler: read_bfee
Please select from 0-0 
Compiler:

zainalabidenakhtar, 2016-05-21:
@haimapi i am using 64-bit MATLAB 7.8.0.347 (R2009a),  (64-bit Window-10) and facing the same issue if u find any solution please guide me

haimapi, 2016-06-12:
@zainalabidenakhtar seems that what it matter is the version of matlab, try 32-bit matlab :-)

zainalabidenakhtar, 2016-06-15:
@haimapi my problem was resolved after using 32-bit windows with 32-bit MATLAB :). i think some extra files are required for 64-bit MATLAB

PigSea, 2016-06-22:
@haimapi @zainalabidenakhtar 
I use matlab2012a (64bit) and visual studio 2010,since the tool only provide the .mexw32 version of read_bfee, when i recompile readbfee.c, i also encount some problems. I slove it by moving all the definitions of variables to the front of the file and get the .mexw64 version sucessfully. Hope it's useful.

tfsanam, 2017-04-12:
@egaebel Can you please tell me what compiler did you use for Matlab R2014a 64bit? Also have you changed anything on the C source code read_bfee.c file?

egaebel, 2017-04-14:
I'm not sure what you mean by what compiler did I use for Matlab R2014a 64 bit. I used that version of matlab to compile and run my code. If I recall correctly I made some minor changes to read_bfee.c but I do not recall specifically.