dhalperi/linux-80211n-csitool-supplementary #121: Issue about packet payloads

libing64, 2015-11-18:
I want to use the packet payload to get the message from the transmitter, so I enable IWL_CONN_RX_MPDU_MSK by 
modprobe iwlwifi connector_log=0x5
then I can get not only the csi data but also my custom packet payload.

Unfortunately, I can only get csi and the packet payload for a few seconds, after that, all the data I received is the packet payload, no csi data is received.

while (1) {
	
	ret = iwl_netlink_recv(sock_fd, &buf, &len);
	
	if (ret == -1)
		exit_program_err(-1, "recv");
	
	if (buf[0] != IWL_CONN_BFEE_NOTIF && buf[0] != IWL_CONN_RX_MPDU) {
		continue;
	}

	if(buf[0] == IWL_CONN_BFEE_NOTIF) //bfee data 
	{
		...
	}else if(buf[0] == IWL_CONN_RX_MPDU)
	{
		...
	}
}

libing64, 2015-11-18:
I am using injection mode on 5GHz

libing64, 2015-11-20:
It seem a problem only under  channel 36, when I changed to other channels, for example 48 or 64, it works well.
Thanks.

closed 2015-11-20 by libing64

dhalperi, 2015-11-21:
I suspect that what happened is that the packets stopped using HT rates and so, as expected, you would only see payloads and not CSI. (You will see payloads independent of what rates the packets are measured at).

swashah, 2016-11-22:
@libing64 : were you able  to log payload in injection mode? also does setting connector_log=0x5 contains any sequence number of packet as i have seen some where it does?

asifhanif1122, 2017-04-25:
@dhalperi Can you please guide about extraction of payload from data file. I have read almost all issues, including 101,  related to custom payload but none of them is specifically telling about payload separation from each received packet. When code==193, what chunk of code will be used in read_bf_file ?