dhalperi/linux-80211n-csitool-supplementary #122: Three antennas hooked up, but only receiving data for 2?

egaebel, 2015-11-20:
Hi, first off, this tool is clearly pretty excellent.

The Intel 5300 card I'm using is hooked up externally, and as such I have direct access to the antennas and where they are placed.

I've set up the driver correctly per the guide and have been able to log CSI for 2 RX antennas. Now I would like to log CSI for 3 RX antennas, but the third antenna is seemingly not being used. I have a 1x2x30 CSI matrix and rssi_a and rssi_b values, but 0 for rssi_c. 

I originally bought two packs of two antennas each, so I swapped out the antenna hooked up to slot 3 in case it was a hardware problem on the antenna side. Now I get readings for rssi_a and rssi_c, still with a 1x2x30 CSI matrix, so it seems like all the antennas are working, it's just that they aren't all being used. I saw it mentioned in this issue: https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/63 that sometimes a third antenna will be ignored if it doesn't provide enough additional information, could this be the problem? Is there some setting I'm missing where I can specify this?

Thanks!

dhalperi, 2015-11-21:
What frequency are you using? It is common that the "third antenna" put in laptops by manufacturers is crappy and, e.g., only works in 2.4 GHz band. As you saw in the previous post, the solution was to move to a different device with better antennas.

You cannot disable this behavior, unfortunately. If the signal measured by the third antenna is so weak, it is simply not worth the energy to include its contribution to packet decoding, so the hardware does not do it.

egaebel, 2015-11-25:
I'm on the 2.4 GHz band. In a few papers I've read that have used this tool they say they've had to use 5 GHz for their experiments, is that because 3 antennas is only supported at 5 GHz?

The antennas I have are 3rd party and support both 2.4 and 5 GHz bands (These are the antennas: http://www.amazon.com/gp/product/B004ZHT2JE?psc=1&redirect=true&ref_=oh_aui_detailpage_o04_s00)

So the signal at the third antenna is only discarded if it's weak? That shouldn't be my problem then, I'm just testing my setup so I'm using my phone's hotspot (with no auth) and my phone is pretty much right next to the antennas.

dpward, 2015-11-25:
My colleagues have suspected that the Intel 5300 adapter experiences crosstalk on the port for the third (center) antenna, as a result of the way the adapter was designed. This would be in addition to any issues with the antennas themselves. @dhalperi please correct me if that's inconsistent with what you have seen.

egaebel, 2015-11-27:
Crosstalk couldn't totally destroy the received signal though right? I'm getting 0 RSSI.

egaebel, 2015-12-03:
Could you post what antennas you have used in your experiments? I can easily switch out antennas to see if that is the problem.

libing64, 2015-12-07:
I am using injection mode, and rssic is not 0, however, the phase of the third antenna(I mean the antenna with smallest rssi) is not accurate. #124

egaebel, 2015-12-07:
@libing64 What equipment are you using?

libing64, 2015-12-07:
@egaebel Just NUC + Intel5300 network card + 3 normal antennas.

egaebel, 2015-12-07:
So, normal laptop antennas? What is your signal source?

libing64, 2015-12-07:
three antennas are put in a row as a linear array.
[image]

What about you?

egaebel, 2015-12-07:
At the moment I am using these flimsy laptop antennas: http://www.amazon.com/gp/product/B004ZHT2JE?psc=1&redirect=true&ref_=oh_aui_detailpage_o04_s00
These were also configured in a linear array.

I had thought that you might be using the better, rotating, screw-in antennas, before your response I ordered three antennas similar to the ones you're using to replace mine.. 

Also my signal source is my phone with the wireless hot spot feature turned on, I'm hoping a switch to a router might help :/

@libing64

dhalperi, 2015-12-09:
@dpward I don't have any proprietary information to that effect, but issues with antenna C in general would not surprise me. It was always much less reliable. and experienced issues.

egaebel, 2015-12-31:
I switched to a router as my signal source and I'm able to pick up all three antennas. I guess the transmit power on my phone was just too weak.

closed 2016-01-02 by dhalperi

CrystalWong2, 2016-09-03:
I have the different problem：two antennas hooked up, but  receiving data for 3? The Intel 5300 card I'm using is hooked up externally, and on the 5GHZ. Did any one kown why?

asifhanif1122, 2016-09-05:
@CrystalWong2  Ignore CSI values for 3rd antenna. You will observe antenna-3 CSI values are smaller in magnitude and will be taken as noise on 3rd slot of Intel 5300 card. If you attach 3rd antenna on slot#3 of wifi card, CSI values will be valid.

CrystalWong2, 2016-09-13:
@asifhanif1122 Thank you for your reply.I think I made a mistake. I used AP as a sender. If I use 2 antenna which connect to the 5300 card，then the Rx should be 2. Am I right?