dhalperi/linux-80211n-csitool-supplementary #123: Unable to set transmit rate in monitor mode

asifhanif1122, 2015-11-21:
Hi, I am having an error in setting transmit rate
sudo echo 0x4101 |sudo tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/iwldvm/debug/monitor_tx_rate
output is:
sudo: tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/iwldvm/debug/monitor_tx_rate : command not found

and for:
sudo echo 0x4101 |sudo tee 'find /sys -name monitor_tx_rate'

output:
find: '/sys/kernal/debug': Permission denied

modinfo iwlwifi | head -n 1 is giving the same name as in /lib/modules/.../updates/ directory. I have seen FAQ and many related issues. Issue is I am not having access to debug folder. I can't see its contents too. Please help. I am using Dell inpiron N4050 as transmitter and Dell inspiron N5110 as receiver

dhalperi, 2015-11-21:
Sounds like you are using the wrong version of the kernel and/or the driver. Please double-check your configuration.

asifhanif1122, 2015-11-22:
My Kernal version is 3.19.0-15.15 and loaded firmware version is 8.24.2.12. What is happening wrong?

dpward, 2015-11-23:
@asifhanif1122 You can't see the contents of `/sys/kernel/debug` unless you have superuser privileges (e.g. by issuing commands with `sudo`). This means for example that with a terminal for a normal user, you cannot use tab-completion to locate files under `/sys/kernel/debug`; you have to actually issue the `sudo` command to list the files, e.g.:
```sudo ls /sys/kernel/debug/```
```sudo ls /sys/kernel/debug/ieee80211/```
etc.

Your first error was `command not found`. Did you type `tee` correctly? If you run the command
`which tee`
does it print the path to the executable, e.g., `/usr/bin/tee`?

Your second error was `Permission denied`, which occurs because `sudo` has to be used separately inside any subshell:
```echo 0x4101 | sudo tee `sudo find /sys -name monitor_tx_rate` ```

asifhanif1122, 2015-11-23:
I was missing sudo but still I am unable to receive any csi packet neither in monitor mode nor using a router.
output of dmesg|grep wifi:

[   14.322194] iwlwifi: unknown parameter 'connector_log' ignored
[   14.483344] iwlwifi 0000:09:00.0: Direct firmware load for iwlwifi-5000-5.ucode failed with error -2
[   14.483351] iwlwifi 0000:09:00.0: request for firmware file 'iwlwifi-5000-5.ucode' failed.
[   14.483372] iwlwifi 0000:09:00.0: Direct firmware load for iwlwifi-5000-4.ucode failed with error -2
[   14.483375] iwlwifi 0000:09:00.0: request for firmware file 'iwlwifi-5000-4.ucode' failed.
[   14.483387] iwlwifi 0000:09:00.0: Direct firmware load for iwlwifi-5000-3.ucode failed with error -2
[   14.483390] iwlwifi 0000:09:00.0: request for firmware file 'iwlwifi-5000-3.ucode' failed.
[   14.650733] iwlwifi 0000:09:00.0: Firmware has old API version, expected v5, got v2.
[   14.650738] iwlwifi 0000:09:00.0: New firmware can be obtained from http://www.intellinuxwireless.org/.
[   14.650875] iwlwifi 0000:09:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[   14.703221] iwlwifi 0000:09:00.0: CONFIG_IWLWIFI_DEBUG disabled
[   14.703226] iwlwifi 0000:09:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[   14.703228] iwlwifi 0000:09:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[   14.703231] iwlwifi 0000:09:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[   14.703454] iwlwifi 0000:09:00.0: L1 Disabled - LTR Disabled
[   15.436380] iwlwifi 0000:09:00.0: L1 Disabled - LTR Disabled
[   15.439239] iwlwifi 0000:09:00.0: Radio type=0x0-0x2-0x0
[   15.583771] iwlwifi 0000:09:00.0: L1 Disabled - LTR Disabled
[   15.586635] iwlwifi 0000:09:00.0: Radio type=0x0-0x2-0x0

I think modified drivers are not being loaded. Where is fault?

asifhanif1122, 2015-11-23:
I have re-installed Ubuntu 14.04.3 LTS with tag=Ubuntu-lts-3.19.0-25.26_14.04.1

ozawali, 2015-11-26:
I recommand you use the 10.04 it is OK

dhalperi, 2015-12-09:
The message that `[ 14.322194] iwlwifi: unknown parameter 'connector_log' ignored means that you did not successfully install the modified kernel and load the modified driver.

Make sure that you have followed the instructions correctly.

Please post the output of `uname -a` and `dmesg | grep -5 -i iwl`

vinatby, 2016-05-18:
I am having the same problem here. Since no body has replied in this discussion from long time, I am going to give outputs of my computer for above mentioned code.  uname -a gives me 
Linux tug14467-Q550LF 4.2.0-35-generic #40~14.04.1-Ubuntu SMP Fri Mar 18 16:37:35 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

and dmesg | grep -5 -i iwl gives

[   29.986694] ACPI Warning: \_SB_.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150619/nsarguments-95)
[   46.666860] audit_printk_skb: 132 callbacks suppressed
[   46.666862] audit: type=1400 audit(1463544821.403:68): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=1952 comm="apparmor_parser"
[   46.666868] audit: type=1400 audit(1463544821.403:69): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=1952 comm="apparmor_parser"
[   46.667120] audit: type=1400 audit(1463544821.403:70): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=1952 comm="apparmor_parser"
[  154.795752] iwlwifi: unknown parameter 'connector_log' ignored
[  154.795904] Intel(R) Wireless WiFi driver for Linux
[  154.795906] Copyright(c) 2003- 2015 Intel Corporation
[  154.796044] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  154.796545] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-5.ucode failed with error -2
[  154.796552] iwlwifi 0000:03:00.0: request for firmware file 'iwlwifi-5000-5.ucode' failed.
[  154.810923] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-4.ucode failed with error -2
[  154.810930] iwlwifi 0000:03:00.0: request for firmware file 'iwlwifi-5000-4.ucode' failed.
[  154.810954] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-3.ucode failed with error -2
[  154.810956] iwlwifi 0000:03:00.0: request for firmware file 'iwlwifi-5000-3.ucode' failed.
[  154.861802] iwlwifi 0000:03:00.0: Firmware has old API version, expected v5, got v2.
[  154.861807] iwlwifi 0000:03:00.0: New firmware can be obtained from http://www.intellinuxwireless.org/.
[  154.861894] iwlwifi 0000:03:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[  154.874704] cfg80211: World regulatory domain updated:
[  154.874707] cfg80211:  DFS Master region: unset
[  154.874709] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  154.874712] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  154.874714] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  154.874715] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm), (N/A)
[  154.874717] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  154.874718] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  154.922083] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[  154.922086] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[  154.922087] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[  154.922089] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[  154.922144] iwlwifi 0000:03:00.0: L1 Enabled - LTR Disabled
[  154.979684] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[  155.049667] cfg80211: Regulatory domain changed to country: US
[  155.049670] cfg80211:  DFS Master region: unset
[  155.049671] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  155.049673] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2700 mBm), (N/A)
[  155.049674] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 1700 mBm), (N/A)
[  155.049675] cfg80211:   (5250000 KHz - 5330000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (0 s)
[  155.049677] cfg80211:   (5490000 KHz - 5600000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (0 s)
[  155.049678] cfg80211:   (5650000 KHz - 5710000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (0 s)
[  155.049679] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 3000 mBm), (N/A)
[  155.049680] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[  155.067647] iwlwifi 0000:03:00.0 wlan1: renamed from wlan0
[  155.087503] systemd-udevd[2126]: renamed network interface wlan0 to wlan1
[  155.088881] iwlwifi 0000:03:00.0: L1 Enabled - LTR Disabled
[  155.092235] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  155.232238] iwlwifi 0000:03:00.0: L1 Enabled - LTR Disabled
[  155.235298] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  155.272522] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  160.625889] wlan1: authenticate with 20:10:7a:9a:d1:e7
[  160.634508] wlan1: send auth to 20:10:7a:9a:d1:e7 (try 1/3)
[  160.645470] wlan1: authenticated
[  160.645584] wlan1: waiting for beacon from 20:10:7a:9a:d1:e7
--
[  315.117194] ACPI Warning: \_SB_.PCI0.RP05.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150619/nsarguments-95)
[  710.964371] [drm:intel_pipe_config_compare [i915]] *ERROR* mismatch in ips_enabled (expected 1, found 0)
[  710.964380] ------------[ cut here ]------------
[  710.964429] WARNING: CPU: 0 PID: 1115 at /build/linux-lts-wily-H6pWeh/linux-lts-wily-4.2.0/drivers/gpu/drm/i915/intel_display.c:12340 check_crtc_state+0x282/0x3f0 [i915]()
[  710.964432] pipe state doesn't match!
[... cut ...]
[ 3004.135667] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[ 3004.273658] iwlwifi 0000:03:00.0: L1 Enabled - LTR Disabled
[ 3004.276754] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[ 3004.306980] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[ 3031.107410] wlan1: authenticate with 20:10:7a:9a:d1:e7
[ 3031.112990] wlan1: send auth to 20:10:7a:9a:d1:e7 (try 1/3)
[ 3031.115421] wlan1: authenticated
[ 3031.118218] wlan1: associate with 20:10:7a:9a:d1:e7 (try 1/3)

I see 'connector_log'  ignored here but i don't think I installed modified kernel ..

closed 2016-08-19 by asifhanif1122