dhalperi/linux-80211n-csitool-supplementary #124: relative phase is not right?

libing64, 2015-11-30:
I want to do some experiments following this paper, “Accurate indoor localization with zero start-up cost”
http://dl.acm.org/citation.cfm?id=2639142
According to this paper, the relative phase between 2 rx antenna should be 
2*pi*r/lambda*cos(alpha)
r is the distance between two rx antennas
lambda is the wave length of the wifi signal
alpha is the relative angle between the antenna array direction and the wifi signal direction.

I have tested the relative phase with only two antennas, it is right and the result just like the math model.

==================================================
However, if I use 3 antennas for receiving, (1 2 3 as a linear array, so the relative phase between 1 and 2,  2 and  3 should be the same in theory ) , the relative phase between 1 and 2 is really different with  the relative phase between 2 and 3.

 
[image]

[image]

libing64, 2015-11-30:
5GHz, injection mode, outdoor environment

libing64, 2015-12-01:
@dhalperi @dpward @ippeb @xu-wang11
Do you have any idea about this problem? Great thanks for your kind help.

dpward, 2015-12-01:
I noticed the same behavior you did, and as I mentioned in #122, the authors of that paper (who I was collaborating with) believed crosstalk was an issue with using the third antenna.

libing64, 2015-12-01:
Do you know how to remove the crosstalk? @dpward

libing64, 2015-12-01:
I don't think it's caused by crosstalk. 
According to wiki, "In wireless communication, crosstalk is often denoted co-channel interference, and is related to adjacent-channel interference"
"Co-channel interference or CCI is crosstalk from two different radio transmitters using the same frequency."

If it is caused by crosstalk, the relative phase will also be affected when I use only two rx antennas. However, when I use only two rx antennas, the relative phase really follows the math model.

x-axis: yaw angle
y-axis: relative phase
Collected data and the math model
[image]
[image]

dpward, 2015-12-01:
@libing64 I was referring to crosstalk in the adapter itself, not wireless communication crosstalk. Also from [Wikipedia](https://en.wikipedia.org/wiki/Crosstalk_(electronics):

"In integrated circuit design, crosstalk normally refers to a signal affecting another nearby signal. Usually the coupling is capacitive, and to the nearest neighbor, but other forms of coupling and effects on signal further away are sometimes important, especially in analog designs."

libing64, 2015-12-01:
@dpward Great thanks.
I find that the relative phase in this paper  "SpotFi: Decimeter Level Localization Using WiFi" is not the same too.  Maybe this paper has solved this problem.
I will try the method used in this paper. 
[image]

lixiaohai, 2015-12-04:
@libing64 
hello!I am also donging some experiment on relative phase.I find that the relative phase  you get change 2pi at some points,it's just like mine.Do you know whether it's normal ?or do I have used the wrong method to get the relative phase?here is one of my data:

[image]

libing64, 2015-12-05:
@lixiaohai I think it is normal to change 2*pi, because alpha and alpha+k*2*pi is the same angle.

lixiaohai, 2015-12-07:
Much appreciation for your kind reply!!

closed 2015-12-09 by dhalperi

lidongmeng, 2016-03-09:
@libing64 hello, I am wondering if u can reimplement the method "SpotFi: Decimeter Level Localization Using WiFi", recently I am reading this paper, but I have lots of question, can u  help me.

libing64, 2016-03-09:
@lidongmeng sorry, I cannot reimplement that method. I once tried that method, but it didn't work on my dataset.

shaozhuchuang, 2016-03-11:
@libing64  I also do an experiment and collect some csi information, but i found the data is very not stable, I don't know whether my data and my matlab code have some problems, Can you send your csi  files and relate matlab code to me, My email is shaozhuchuang@126.com or mrshao@sjtu.edu.cn. Thank you!

egaebel, 2016-03-16:
It seems that many of us are interested in SpotFi. @libing64 were you able to correctly implement Algorithm1 from the SpotFi paper to remove the offset from sampling time? It seems straightforward, but my implementation does not give results similar to the one from the paper.

libing64, 2016-03-17:
@egaebel It's very simple to remove the phase offset.
Look at the phase picture,  and if you add some codes like this, 
if( (phase_current - phase_prev) > pi*1.5 ) phase_current -= 2*pi
if( (phase_current - phase_prev) < -pi*1.5 ) phase_current += 2*pi
then it will be similar to the figure from SpotFi. 
https://cloud.githubusercontent.com/assets/3192355/11477501/45a66da0-97c3-11e5-8448-052d4e575520.png

egaebel, 2016-03-17:
@libing64 wait, how does that implement SpotFi's sanitization algorithm?

[image]

libing64, 2016-03-17:
@egaebel It's just a simple line fitting, assume that phase  = a*t + b
then the phase without offset is phase_without_offset = phase - a*t

egaebel, 2016-03-17:
@libing64 Just to be clear, is phase_prev the phase of the previous subcarrier on the same packet, or the phase of the same subcarrier on the previous packet?

Thank you so much for helping me with this.

libing64, 2016-03-17:
@egaebel 
It is the phase of the same subcarrier on the previous packet

cedivad, 2016-04-11:
@libing64, sorry for bothering you, but I'm asking since that it looks like you understood it. What is beta in the first formula of the first algorithm for spotifi? I think I got pretty close to have it working, but only when I multiply the STO by say 2 to 3 before using it to correct the phase. I believe it's because I didn't use beta in the formula searching for STO, because I didn't knew what to put in.

I belive it must be something so evident to everyone (but me) the authors didn't even bother specifying. 

Thank you.

shaozhuchuang, 2016-04-18:
@egaebel Are you implement SpotFi's sanitization algorithm? I don't know how to do . can you help me? thank you!

shaozhuchuang, 2016-04-18:
@libing64 I found we can get unwrapped phase bu using unwrap command in matlab, I don't the difference with you method mentioned above, can you help me about your method get unwrapped phase. i need your help.thank you!

libing64, 2016-04-18:
@shaozhuchuang  It seems same as SpotFI. What you mean by "this phase is different with spotf“?

shaozhuchuang, 2016-04-19:
@libing64  According to spotfi, the phase difference of different packages is bigger than the phase difference of different antenna using the same package, but i don't find it in my picture.

libing64, 2016-04-19:
@shaozhuchuang Please note that the phase difference between two antennas is related to the distance between two antennas.

asifhanif1122, 2016-04-19:
@libing64 Can you kindly explain SpotFi's alogorithm-1 . I am unable to understand it. Especially the terms used in it. Ro and veeta . Which parameters are they corresponding to? 

[image]

cedivad, 2016-04-19:
p is the term you need to minimise (arg min of p), I think you can leave B at zero.

egaebel, 2016-04-19:
p (rho) and B (beta) are terms of a linear fit.

cedivad, 2016-04-19:
@egaebel I'm guessing, p is the slope, B the intercept? I looked at the Wikipedia page for the beta parameter before but I really couldn't get it, so I kept it to zero and my results weren't that bad (I couldn't get past the most basic MUSIC graph, but I was expecting that). I really regret lacking basic math skills here.

egaebel, 2016-04-19:
Yep. You got it. Just dig through the math and keep digging until you understand. I had to do the same thing several months ago. It all clicks eventually :).

shaozhuchuang, 2016-04-20:
@cedivad @egaebel Are you implement SpotFi's sanitization algorithm? can you share it with me . I have studied for a long time. Thank you!

libing64, 2016-04-20:
@shaozhuchuang I tried to implement the algorithm in SpotFi several month's ago, but it doesn't work. Please note the evaluation in SpotFi. It seems that  it doesn't works much better than general MUSIC algorithm for AoA estimation.
SpotFi’s accuracy stems from two key components. First is its super-resolution AoA estimation algorithm whose
**median error is less than 5 degrees in LoS conditions and less than 10 degrees in NLoS conditions. The MUSIC al- gorithm used in practical implementation of ArrayTrack only achieves an estimation error of 7.4 and 15.2 degrees respectively for the same scenarios**. Second, SpotFi’s likelihood estimation algorithm ensures that among the available estimates, incorrect estimates for the direct path are likely not picked, and even if they are, they will be weighted with low confidence and have a lesser impact on the location estimate

shaozhuchuang, 2016-04-20:
@libing64 thank you for your reply! Are you implement super-resolution AoA estimation algorithm? i only know how to implement basic music algorithm to calculate AoA , and i don't know how to combine AoA and ToF in music algorithm. besides,i don't use subcarrier information to calculate AoA. Can you give me some instructions? Thank you!

MBrend15, 2016-04-22:
Great comments on SpotFi! 

This is more of a MATLAB specific question. How does one calculate regression over all antennas/subcarrier? Basically I used the mvregress command (below, b is a 1x30 column vector of values, from 1-30, for x values, then phase_mat is a 3x30 array) and then took the average of the three returned coefficients (Each coefficient only represented a regression over a single antenna). This returns a workable result but I know this is not the most accurate way to find best fit over values of each antenna and sub-carrier. 

y = mean(mvregress(b,phase_mat));

Thanks for the help!

egaebel, 2016-04-22:
I have also implemented SpotFi, although only from a single point, there is no result fusion between access points. I've found it to perform pretty well, although not quite as well as I had expected. I can post my code here once I open it up to the public in 2-3 weeks.

egaebel, 2016-04-22:
@shaozhuchuang I had to read the SpotFi paper many, many times before I could properly understand how to combine ToF and AoA in MUSIC. Examine their construction in figure 3. Then examine it over and over again until it makes sense :). It's important to understand that MUSIC works on sensors, not antennas. What the authors of SpotFi demonstrate in Figure 3 is how to reconstruct the ToF and AoA data obtained from antennas into many sensors.

[image]

libing64, 2016-04-23:
@egaebel  Good new that someone successfully implement SpotFi algorithm. I once failed to re-implement SpotFi, but it didn't work with my own data(indoor environment).   (Maybe there are some bugs in my code)

Are you using Intel 5300 NIC or Atheros NIC for collecting CSI data? 
Is the data collected indoor or outdoor?

shaozhuchuang, 2016-04-24:
@egaebel  Great!! I',m so happy to your implemention, Looking forward for your code and result! I thin it can help me greatly!  Thank you!

dongxinb, 2016-04-25:
@egaebel @libing64 

Sorry to be a bother. These days I'm trying to implement the SpotFi and here is a question about CSI Smoothing.
As shown in the image, the size of the matrix is (30*30). But the actual size in this matrix is (30*32) because there are 32 columns and 30 rows. 
I'm not sure this is just a typo or something else. Could you give me some instructions?
Thanks a lot!

[image]

egaebel, 2016-04-26:
@dongxinb 

No problem! I actually e-mailed the authors about this and they told me that the matrix is 30x32. The 30x30 indication is a typo.

egaebel, 2016-04-26:
@libing64 I'm using the Intel 5300 NIC with indoor data. I was going to gather outdoor data too (and probably will eventually) but I've been strapped for time. I'm running SpotFi from a single access point so I'm only obtaining the angle of arrival and it is only from the one point, there's no results fusion. My results are not too bad. I find that it's more useful to look at the results in terms of top-5 accuracy because of the likelihood selection method. Oftentimes the wrong cluster is selected, but not by much. From 1m away I get 100% top-4 accuracy.

I'm using this technique for my masters thesis, I can post a final paper here covering it when I'm all done :).

libing64, 2016-04-26:
@egaebel Great thanks for your reply.
I will pay more time to debug my code.

dongxinb, 2016-04-26:
@egaebel 
Thank you for you kind reply.
I'll tell my thoughts below and could you please evaluate if my thought is correct?

1. Collect the data from csitool and get the matrix.

2. Use the Algorithm 1 to sanitize the ToF estimates. 
2.1. I convert the complex to angle and module, and remove the periodic effect. (Make the points look like a line rather than periodic polyline).
2.2. Get the linear fitting result(y-axis is phase, x-axis is different subcarriers in one packet), the slope is p. Then new_phase_of_subcarrier_i=phase_original_of_subcarrier_i - i * p.
2.3 Using the new phase and reconstruct the csi_data_matrix.

3. Construct the smoothing matrix using the new csi_data_matrix. The size is 30x32.
3.1. Using the matlab function "pmusic(matrix, 30)" to get the result: s, w.
3.2. Find the max(s) and its index, then w(index) is the unknown angle.

Besides, could you share your csi_data file and the real angle(if possible)? I think my data might have some problems, and I cannot capture new data these days.
Thanks again!

plank3ffect, 2016-04-26:
@egaebel @dongxinb ,

My understanding for the MUSIC algorithm is that you really have to write your own according to the mathematical model presented as a combination of RSSI_MN = Phi_theta_k^M * Omega_tao_k^N  (Eqn 1 x Eqn 6).  

@egaebel
How exactly did you implement the MUSIC algorithm?  Did you create a separate MUSIC algorithm problem for each path, set up similar to Figure 3b?  Or did you somehow operate on the entire [30x32] smoothing matrix?  My thoughts are that for every path, you have to model the following:
[choose two 30x1 subarrays to compare] = [corresponding 30x2 steering] x [2x2 independent vector of gains].  Any help or comments on your approach would be greatly appreciated.

@dongxinb, I found this website really helpful in understanding composing the MUSIC algorithm!
http://www.edaboard.com/thread79938.html

egaebel, 2016-04-26:
@dongxinb @plank3ffect 

You need to roll your own MUSIC implementation. pmusic in matlab is not equipped to handle time of flight. SpotFi builds directly on top of MUSIC, fundamentally altering it so to be able to do this you need your own implementation.

@plank3ffect 
I'm not quite certain what you mean by "a separate MUSIC algorithm problem for each path." Figure 3b is an much smaller explanation of how to construct the smoothed CSI matrix in general. The authors move certain terms into the matrix F from music so that the matrix A can be constructed to only be made up of steering vectors. It's sort of like a matrix factorization, but a very creative one. 

The 30 x 32 smoothed matrix is the signal matrix that is input to MUSIC. You find its eigenvectors and construct E_{N} which is finally multiplied in the music peaks equation P_{MUSIC}(\theta, \tau) = 1 / a^{H}(\theta, \tau) E_{N} E^{H}_{N} a(\theta, \tau).

Figure 3b is purely for explanatory purposes.

This paper was crucial to my understanding of MUSIC: http://www.diva-portal.org/smash/get/diva2:724272/FULLTEXT01.pdf

plank3ffect, 2016-04-27:
@egaebel ,
Thanks for the feedback!  Can you shed some light on your search space with only the single smoothed matrix as the objective measurement matrix?  My intuition is that one would have to search through (\theta_{k},\tau_{k}) with k \in [1,5].  In other words, an optimization problem with 10 variables,  \theta and \tau for each path, assuming five paths.  That would take a long time! (right?)

egaebel, 2016-04-27:
@plank3ffect 
You could make this an optimization problem I suppose....or you can just plug in combinations of values for\ tau and \theta based off of what we know. For angle of arrival I try every angle from -90 to 90 and for time of flight I use...I think microsecond granularity and go up to a few hundred microseconds. I'll double check this later, but the point is that treating this as an optimization problem from the outset without using the prior information we have about the problem is not a good starting point.

plank3ffect, 2016-04-27:
@egaebel,

I think we are getting to the same page of understanding.  I have it set up to search every angle for AoA and time for ToF, but that is for only one path.  From what I gather, each path has a different AoA and ToF.   The SpotFi paper mentions they analyzed for five paths, and it seems the smooth matrix then incorporates all paths.  That's what threw me off.  Were you able to extract the variable pairs (AoA, ToF)_path1, (AoA, ToF)_path2, (AoA, ToF)_path3, (AoA, ToF)_path4, (AoA, ToF)_path5 from a single CSI matrix?   Does that mean you swept through 10 variables over this one matrix?  Trying to see if I'm framing this right for five paths.  Or did you only solve for one path?  THanks!

egaebel, 2016-04-27:
@plank3ffect 

You should get as many peaks in the music spectrum as there are paths. So if there are 5 paths then you should expect to see 5 peaks in the music power spectrum. Ideally, the largest of these will be the direct path. There's variability though, which is why the SpotFi authors compute this for each packet and then cluster all the packets.

shaozhuchuang, 2016-05-05:
@egaebel Are you know how to get the likelihood of each cluster, please tell me if you know about it.Thank you!

ColdCodeCool, 2016-05-05:
@shaozhuchuang  Hi, there! I recognized you are from SJTU,  so we are both in Shanghai. I am a first year Ph.d student in the newly built ShanghaiTech.  I've sent you an email to your school mailbox, please check it, hope to talk with you soon!!
Regards

MBrend15, 2016-05-05:
@shaozhuchuang 

The authors of the paper indicated that to determine likelihood score, you have to weigh your variation of AoA, ToF, number of packets per cluster, and average ToF with weight values but they do not include the weights in the work. I took that to mean that they can be arbitrary and chose values that returned a reasonable likelihood score per cluster (usually between 1 or 2). Obviously this is my interpretation and I would be interested to get the group's opinion.

egaebel, 2016-05-06:
I'm not sure how the authors actually determined the likelihood weights.

I used a linear SVM on labeled data to determine them.

egaebel, 2016-05-10:
As promised, here is a link to my SpotFi implementation:

https://github.com/egaebel/lgtm/tree/master/csi-code

In case the links change in the future the master project is here:

https://github.com/egaebel/lgtm/

ColdCodeCool, 2016-05-11:
@egaebel  I got a naive question here, how should we obtain the ground truth incident angles of the antenna arrays?  More accurately, we should choose whether the horizontal direction or vertical direction of the antenna arrays as the reference direction to construct the incident angle,  or neither.

Thank you in advance!

lingyan0222, 2016-05-23:
@egaebel 
Hello, thank you for sharing your code for us, first. I have issues about CSI data collection.

When I follow to reimplement SpotFI, I confused that the CSI data collection form both receiver and transmitter must be with Intel 5300. 

The issue is that whether to make 5300 NIC in injection mode to get CSI data for phase processing, or with hostaapd? other AP is OK? How do you get the CSI data? Thank you a million.

egaebel, 2016-05-23:
@ColdCodeCool
Use the horizontal direction, the long one that is.

@lingyan0222 
You should be able to get CSI data in, I think, any of the modes. For SpotFi I used injection mode since my particular application was better suited for it. Only way to figure it out for yourself is to dive in and try.

lingyan0222, 2016-05-24:
@egaebel 
Thank you for your prompt feedback. I will try it.

ColdCodeCool, 2016-05-24:
@egaebel 
Thank you for your kind reply.

In your code, at the sanitization part, I think fit_X(1:30,1) = 0:1:29, and so are fit_X(31:60,1),fit_X(61:90,1).

And fit_X = fit_X * 2 * pi * delta_f;

Then the polyfit will do it right. Do you agree with me?

libing64, 2016-05-25:
@egaebel 
I compared the logged data from you and me, it seems that the phase of the third antenna is right in your data but not in my logged data.
This is the from your logged file. Note that the relative phases really fit the math model.
[image]

This is from my logged data.
[image]

In my experiment, when I uses two antennas, the relative phase fits the math model well, but when I use three antennas, the phase from the third antenna is always very strange. 
Are you using injection/monitor mode or others? Can you show me your configure file at transmitter and receiver? Great thanks.

[28 more replies]