dhalperi/linux-80211n-csitool-supplementary #125: Can't connect to unencrypted AP route after reloading driver

liujunyi-sjtu, 2015-12-02:
Dear all, I've reboot the computer after installation.
Before I unload the driver, I can connect to an unencrypted TP-LINK AP route(unencrypted, 802.11n, 2.4Ghz) with network-manager and I can't connect to any encrypted AP at the same time, so I think the firmware may has been installed well. 
But after executing these two commands 
```
sudo modprobe -r iwlwifi mac80211
sudo modprobe iwlwifi connector_log=0x1
```
I can't connect to the unencrypted AP route any more. Does CSI Tool has any specific requirement for the AP that is connected besides unencrypted and 802.11n?
I'm using Ubuntu 14.04.2 with kernel 3.16.0.
[  275.850549] cfg80211: Calling CRDA to update world regulatory domain
[  275.851965] Intel(R) Wireless WiFi driver for Linux, in-tree:
[  275.851968] Copyright(c) 2003- 2014 Intel Corporation
[  275.852142] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  275.852400] iwlwifi 0000:03:00.0: irq 45 for MSI/MSI-X
[  275.853573] cfg80211: World regulatory domain updated:
[  275.853577] cfg80211:  DFS Master region: unset
[  275.853578] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  275.853581] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  275.853584] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  275.853586] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm), (N/A)
[  275.853588] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  275.853590] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
[  275.853689] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[  275.853692] iwlwifi 0000:03:00.0: Falling back to user helper
[  275.854422] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[  275.854425] iwlwifi 0000:03:00.0: Falling back to user helper
[  275.854803] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[  275.854806] iwlwifi 0000:03:00.0: Falling back to user helper
[  275.856313] iwlwifi 0000:03:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
--
[  275.866210] iwlagn: connector callback registered
[  275.866227] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[  275.866228] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[  275.866229] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[  275.866231] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[  275.866444] iwlwifi 0000:03:00.0: L1 Enabled; Disabling L0S
[  275.891439] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[  275.912355] iwlwifi 0000:03:00.0: L1 Enabled; Disabling L0S
[  275.915470] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  276.054292] iwlwifi 0000:03:00.0: L1 Enabled; Disabling L0S
[  276.057329] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  276.096863] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

[  331.578401] wlan0: authenticate with 06:11:b5:1f:e9:16
[  331.580305] wlan0: send auth to 06:11:b5:1f:e9:16 (try 1/3)
[  331.584962] wlan0: authenticated
[  335.686191] wlan0: aborting authentication with 06:11:b5:1f:e9:16 by local choice (Reason: 3=DEAUTH_LEAVING)
--
[  345.724229] wlan0: authenticate with 12:11:b5:1f:e9:16
[  345.733307] wlan0: send auth to 12:11:b5:1f:e9:16 (try 1/3)
[  345.751792] wlan0: authenticated
[  350.740948] wlan0: aborting authentication with 12:11:b5:1f:e9:16 by local choice (Reason: 3=DEAUTH_LEAVING)
--

liujunyi-sjtu, 2015-12-02:
I've solved the problem.
I used iw to connect the AP route and set the ip for wlan0 then log_to_file worked.

zhangshaohu, 2016-04-18:
Junyi, I have a similar problem with you. could you tell me which command do you use? sorry I am green for Linux

ajebulon, 2016-07-19:
Hi Junyi, could you please elaborate how can you fix it? I'm facing the same problem with you. I 've already finished step 1-4 (I skipped the blacklist command in the step 1), and now when I tried doing the step 5, I couldn't connect to any AP, either encrypted or unencrypted one. I'm using Ubuntu 14.04.4 LTS.

I also have tried using the iw command to connect to my AP:
`iw wlan0 connect {my SSID}`
, but it does nothing. No output or whatsoever.

Any suggestions anyone?

liujunyi-sjtu, 2016-07-19:
@ajebulon 
You may set the IP of your client by
`ifconfig wlan0 {arbitrary IP in the segment of AP such as 192.168.1.123 if your AP's segment is 192.168.1.x}`
before connecting the AP.
DHCP seems not working on my client with CSI Tool.

ajebulon, 2016-07-20:
@liujunyi-sjtu 
Seems like we have different cause of problems. I've already solved mine by:
1. Doing the blacklist part mentioned in the step 1
2. Restart the computer -> No need to worry about seeing the WiFi adapter not working at this stage
3. Do the step 5

The "driver unloading" in step 5 resulted in 'module not found' or something like that. And after reloading the driver CSI logging enabled, my WiFi could connect again to the unencrypted AP.

tahmidzbr, 2016-10-18:
@liujunyi-sjtu @zhangshaohu @ajebulon @dpward @dhalperi  can someone please put some light on how to log_to_file csi.dat? I know my installations are correct after reading other posts in GitHub, but when i ping to my AP, i dont see any CSI being measured.

My AP is set to work at n mode.