dhalperi/linux-80211n-csitool-supplementary #126: set transmission rate to 10pkts/s

star90, 2015-12-02:
Hi guys,

I am wondering is that possible to set transmission rate to 10pkts/s when using a 5300 card and a commercial AP?  If yes, how to set this?

Regards

dpward, 2015-12-03:
If you are using the `ping` command, for example, you can add the `-i` flag to set the time interval between packets sent:
`ping -i 0.1`

Another option is to use traffic generation tools such as [mgen](http://www.nrl.navy.mil/itd/ncs/products/mgen) which allow setting this interval and much more.

star90, 2015-12-03:
@dpward Thank you very much for your reply! I will try this.

star90, 2015-12-03:
Hello, I have used the method you said to set the transmission rate to 10pkts/s. However, in the terminal used to ping, it said 136 packets were transmitted, but why I captured 149 packets in the CSI file? Shouldn't these two numbers the same?

I am using a commercial router as the AP. Does this have a problem?

Looking forward to your reply.

Regards

dpward, 2015-12-03:
CSI will be collected for any packets that meet the requirements in [FAQ 5](http://dhalperi.github.io/linux-80211n-csitool/faq.html). This includes network control traffic such as ARP, IPv6 Neighbor Discovery, mDNS, etc. aside from the traffic you are generating.

Also consider that if a packet is retransmitted by AP because it did not receive an 802.11 acknowledgement from the receiver, but the receiver did successfully decode both transmission attempts, then there will be two CSI measurements at the receiver.

star90, 2015-12-03:
@dpward 
Thank you very much for very fast reply.
So in this case, if that is possible I receive the same number of packets transmitted by the commercial AP in receiver side? I notice that you mentioned somewhere using 5300 cards for both sides. Can I achieve this by using two 5300 cards?

The reason I want to receive the same number of packets transmitted by the AP and set the transmission rate, is that I want to know the time information from CSI, such that I can do some signal processing. I don't know if there are any methods to have time information. Could you please kindly give me some advices?

Many thanks for your help.

Best wishes

star90, 2015-12-08:
@dpward  
@dhalperi 
Could you please kindly give me some advice？If that is possible I receive the same number of packets transmitted by the commercial AP in receiver side?

Best wishes

dhalperi, 2015-12-09:
Each received packet has an associated timestamp using microsecond information; this should be good enough. (See FAQ.1)

You can use injection mode for 5300-to-5300 communication.

It is very unlikely that you will receive exactly the same number of packets transmitted by the commercial AP. For example, there will be broadcast packets, packets transmitted to other clients, and packet loss.

star90, 2015-12-09:
@dhalperi 
Hi, do you mean the 'timestamp_low' in csi_entry in FAQ.2?

star90, 2015-12-10:
@dhalperi 
Dear Daniel,

I have used the timestamp in each packet to obtain the time information, and I plot a figure like:
[image]

But the problem is, the time interval between adjacent data point are different, which is not suitable for signal processing. I think this is because the received packets are more than transmitted using ping command. 

Do you have any suggestion to have time information with equivalent interval between each other?

Regards,

dhalperi, 2015-12-12:
I think your goal of getting perfect inter-packet timing is hopeless. There is too much delay in the various system components. There must be signal processing techniques that can handle samples with intermittent and slightly variable timings you can find for your purposes.

I could imagine trying to send packets as fast as possible and then picking the packet that came as close as possible to hitting exactly some interval 100*sampling rate apart. E.g., you might try to send super-short packets approximately 100µs apart, and then picking the CSI measurement that comes closest to say an even 10ms sampling rate.

yi-gao, 2015-12-15:
I have a related problem about the timestamps.
I am using to 5300 cards, 5G, monitor mode
The transmitter sends about 1000 pkts/s.
Then at the receiver side, sometimes the timestamps are normal, but sometimes they are not.
For example, when the timestamps are not normal, they look like the following (mod 1e7):
...
5109376
5109376
5111424
5111424
5113472
...
Note that the difference between two different timestamps is always 2048, which cannot be correct.

This behavior seems quite random in my experiments, so I could not find the root cause. Has anyone observed similar behaviors?

star90, 2015-12-16:
@dhalperi 
Hi, many thanks for your reply and suggestion. I think this maybe work, and I will try this.

Regards

star90, 2015-12-16:
@yi-gao 
Hi, could I ask you a question? How do you deal with the two packets that have the same timestamps? For example 5109376 and 5109376 in your comments?

Thanks

yi-gao, 2015-12-16:
@star90 I am also looking for solutions.

Darwintam, 2015-12-17:
@yi-gao  you improved the sampling rate by "ping <AP IP> -i 0.001"?

yi-gao, 2015-12-17:
@Darwintam I am using the monitor mode (https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection)

Darwintam, 2015-12-18:
@yi-gao Thanks for your reply. Could you please tell me what mean"echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate`   when i input this command ,it is said"tee: find /sys -name monitor_tx_rate:could not find this folder or directory 0x4101"

@dhalperi  @yi-gao Help!!!!!   THANKS!!!!

dhalperi, 2015-12-22:
This means that either you are not using the modified driver or you did not run the setup_inject scripts.

metahys, 2016-01-11:
@yi-gao We have the same problem with you. Did you get the solution now?

yi-gao, 2016-01-11:
@metahys Not yet.

lidongmeng, 2016-01-26:
@star90 Hello, I encountered the same problem as yours when I used the commercial AP. I am wondering if we can get the MAC Address of the packet. Have u solved the problem? Thank u!

azizshahics, 2016-06-01:
@ star90
                can you please indicate how to plot csi data in time domain?

asifhanif1122, 2016-06-01:
If you take Ifft of one packet , it will look like this graph. In time domain we call it Power Delay profile.
[image]

azizshahics, 2016-06-01:
@ Asif Hanif, where I can get the packet in matlab? can you please indicate?

azizshahics, 2016-06-01:
Can you please send me the snap of matlab for taking ifft of one packet?

zainalabidenakhtar, 2016-06-21:
@star90 can you please explain how to use time stamp to plot this because i am getting a large amount of time stamp for example 'timestamp_low: 440063326' how to use this for plot?

tahmidzbr, 2016-11-21:
@zainalabidenakhtar Hi could you figure how to use timestamp_low to plot the data in time domain?

ppenumarthi, 2017-04-03:
"If you take Ifft of one packet , it will look like this graph. In time domain we call it Power Delay profile or Channel Impulse Response (CIR)."

@asifhanif1122: 
Do you mean to say: IFFT of the result from "get_scaled_csi(csi_entry);" or directly "csi_entry"? These variables are picked from [FAQ2]( http://dhalperi.github.io/linux-80211n-csitool/faq.html). I get different values every time I run IFFT, main problem is: I still get 30 different values, but don't see TIME in the IFFT result.

ppenumarthi, 2017-04-11:
Do we get multiple CIR (like CIR for each antenna)? Or only one?I seem to be performing IFFT on CSI at each antenna.