dhalperi/linux-80211n-csitool-supplementary #127: Using injection scripts.  FATAL: Module iwlwifi is in use.

star90, 2015-12-03:
Hi Guys,

I am trying to use machine with 5300 card as both transmitter and receiver, and using injection scripts.
However, when I try to run: ./setup_inject.sh 64 HT20, it says 'FATAL: Module iwlwifi is in use'. Is this normal?

Regards

dpward, 2015-12-08:
That script tries to unload and then re-load the driver. On distributions other than Ubuntu, the 'iwldvm' module has to be unloaded before the 'iwlwifi' module is unloaded. (Ubuntu ships with a modprobe alias to do this.) You may want to edit the script as necessary for your purposes.

star90, 2015-12-08:
Thank you very much for your answer. I will as you suggested

closed 2015-12-09 by dhalperi

lidongning, 2017-06-21:
my pc is ubuntu12.04  ,when I run: ./setup_inject.sh 64 HT20,  I also "FATAL: Module iwlwifi is in use" , I don't understand it .  how to edit it ?  @dhalperi

lidongning, 2017-06-21:
my pc is ubuntu12.04 ,when I run: ./setup_inject.sh 64 HT20, I also "FATAL: Module iwlwifi is in use" , I don't understand it . how to edit it ? @dhalperi  @star90