dhalperi/linux-80211n-csitool-supplementary #128: Intel(5300) Wi-Fi Card using Modified Drivers and Firmware---Monitor Mode Setting

asifhanif1122, 2015-12-04:
I am trying to set my Intel wi-fi card in Monitor Mode in Linux. While trying to do this task i run some commands that should  prompt me to enter my GitHub account user ID and password. But instead it shows an error that is {   fatal: Not a git repository(or any of the parent directories):  .git }

These are the commands I am trying to use......................

uname -r

UBUNTU_KERNEL_TAG=Ubuntu-yourKernalTag

UBUNTU_KERNEL_TAG=Ubuntu-3.19.0-1515

(Error appears after these commands)
. /etc/lsb-release
git remote add ubuntu git://kernel.ubuntu.com/ubuntu/ubuntu-${DISTRIB_CODENAME}.git
git pull --no-edit ubuntu ${UBUNTU_KERNEL_TAG}

What should i do to get rid of these errors?

dpward, 2015-12-08:
You should not be prompted for, or enter, a Github username/password when following any of these instructions; you are cloning the repository anonymously.

The error you are seeing indicates that you didn't successfully clone the repository and change into that directory. Try starting at step 2 again and running the commands exactly as shown. If there is an existing directory named `linux-80211n-csitool`, then delete it first.

asifhanif1122, 2016-01-06:
I removed linux-80211n-cistool file and started its downloading again. It went fine but when i tried these commands, error appeared again. Actually my group-fellow was working on this project and she left a document with specific commands for Monitor mode configuration on Linux. Therefore i am facing troubles in this process. According to the description in notes the commands on Terminal window (shown in attached snapshot) should prompt me to enter GitHub user id and password but instead fatal errors appear. I have attached a snapshot of my Terminal window with comment to show error. What mistake am i doing or please suggest me something else that i should try. Thank you. 

[image]

ANeeK181, 2016-06-20:
Instead of . /etc/lsb-release do cat /etc/lsb-release
it will show you some information in which will be your ubuntu version codename, use it instead of variable ${DISTRIB_CODENAME}