dhalperi/linux-80211n-csitool-supplementary #129: problem with building and installing the modified wireless driver

stuartgolden, 2015-12-06:
I am having problems with installing the firmware. I am running into issues with both the old instructions and the new instruction versions. The furthest I got was with the old instructions V2. I am using Ubuntu 15.10, and I received an error at the "make -j3 bzImage modules" step.

At this step, I get an error ...

include/linux/compiler-gcc.h:100:30: fatal error: linux/compiler-gcc5.h: No such file or directory
compilation terminated.

Any thoughts? Also, I am OK with downloading a different version of the OS if that will help.

Thanks for any help you can give.

dpward, 2015-12-07:
The old instructions are only applicable to Ubuntu 10.04 LTS (which has reached end-of-life).

I need to merge/test the modified driver with kernel 4.2 (which Ubuntu 15.10 ships with) and 4.3 &mdash; it's on my to-do list. Ubuntu 14.04.3 LTS, or any regular release through Ubuntu 15.04 should work at the moment.

stuartgolden, 2015-12-08:
> 15.10 ships with) and 4.3 — it's on my to-do list. Ubuntu 14.04.3 LTS, or

Hi David,
    Thanks for the reply. I have now installed ubuntu 14.04.3 lts and I
think I got the driver installed. I am now working on getting the logging
to work.
Regards,
Stuart

stuartgolden, 2015-12-09:
David,
    I ran through the installation instructions and everything worked
exactly as stated until I loaded the new driver. After that the network
manager can not establish a connection. It sees the AP, but fails on
establishing a connection. Any thoughts?
Thanks,
Stuart

vinatby, 2016-05-19:
perhaps your access point information is encrypted... make it unencrypted ..
FAQs.