dhalperi/linux-80211n-csitool-supplementary #131: The csi.dat is always empty? All commands have been executed in Ubuntu 14.04

jiyiren, 2015-12-09:
I use iw to connect the wifi and ping command is ok. My order is following:
sudo modprobe -r iwlwifi mac80211
sudo modprobe iwlwifi connector_log=0x1
sudo iw wlan0 connect jylab
sudo dhcpcd wlan0
sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat

That is all.The terminal didnot show the received data info.I cut a picture whici is:

http://7xknpe.com1.z0.glb.clouddn.com/iwintel5300.png

please give me some suggestions?!! Thanks

dhalperi, 2015-12-09:
Please post the output of the following commands:

* `uname -a`
* `iw dev wlan0 link`
* `iw dev wlan0 scan`
* `dmesg | grep -5 -i iwl`

tahmidzbr, 2016-10-17:
@dhalperi hi Daniel, I have the same problem as @jiyiren.
I will post below the output to your questions

uname -a
Linux lap2-tahmid 3.13.0-95-generic #142-Ubuntu SMP Fri Aug 12 17:00:09 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
------------------------------------------------------------------------------------------------------------------------
iw dev wlan0 link
Connected to 20:a9:d3:7c:39:18 (on wlan0)
	SSID: tahmidzbr
	freq: 2417
	RX: 1853931 bytes (18203 packets)
	TX: 46906 bytes (387 packets)
	signal: -21 dBm
	tx bitrate: 26.0 MBit/s MCS 3

	bss flags:	short-preamble short-slot-time
	dtim period:	0
	beacon int:	100
--------------------------------------------------------------------------------------------------------------------------------
dmesg | grep -5 -i iwl

[   14.144871] yenta_cardbus 0000:03:01.0: pcmcia: parent PCI bridge window: [mem 0xf0000000-0xf3ffffff pref]
[   14.144874] pcmcia_socket pcmcia_socket0: cs: memory probe 0xf0000000-0xf3ffffff:
[   14.144882]  excluding 0xf0000000-0xf3ffffff
[   14.202325] Intel(R) Wireless WiFi driver for Linux, in-tree:
[   14.202329] Copyright(c) 2003-2013 Intel Corporation
[   14.202464] iwlwifi 0000:0c:00.0: can't disable ASPM; OS doesn't have ASPM control
[   14.202526] iwlwifi 0000:0c:00.0: irq 46 for MSI/MSI-X
[   14.278993] kvm: disabled by bios
[   14.292990] kvm: disabled by bios
[   14.326536] [drm] Memory usable by graphics device = 2048M
[   14.326541] checking generic (e0000000 300000) vs hw (e0000000 10000000)
[   14.326543] fb: conflicting fb hw usage inteldrmfb vs VESA VGA - removing generic driver
--
[   14.347424] dcdbas dcdbas: Dell Systems Management Base Driver (version 5.6.0-3.2)
[   14.372083] i915 0000:00:02.0: irq 47 for MSI/MSI-X
[   14.372096] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   14.372098] [drm] Driver supports precise vblank timestamp query.
[   14.372163] vgaarb: device changed decodes: PCI:0000:00:02.0,olddecodes=io+mem,decodes=io+mem:owns=io+mem
[   14.473145] iwlwifi 0000:0c:00.0: Direct firmware load failed with error -2
[   14.473151] iwlwifi 0000:0c:00.0: Falling back to user helper
[   14.650094] input: Dell WMI hotkeys as /devices/virtual/input/input8
[   14.663778] iwlwifi 0000:0c:00.0: request for firmware file 'iwlwifi-5000-5.ucode' failed.
[   14.663810] iwlwifi 0000:0c:00.0: Direct firmware load failed with error -2
[   14.663812] iwlwifi 0000:0c:00.0: Falling back to user helper
[   14.666486] systemd-udevd[576]: Error calling EVIOCSKEYCODE: Invalid argument
[   14.687524] gpio_ich: GPIO from 195 to 255 on gpio_ich
[   14.688519] iwlwifi 0000:0c:00.0: request for firmware file 'iwlwifi-5000-4.ucode' failed.
[   14.688547] iwlwifi 0000:0c:00.0: Direct firmware load failed with error -2
[   14.688549] iwlwifi 0000:0c:00.0: Falling back to user helper
[   14.690255] iwlwifi 0000:0c:00.0: request for firmware file 'iwlwifi-5000-3.ucode' failed.
[   14.735930] fbcon: inteldrmfb (fb0) is primary device
[   14.844424] pcmcia_socket pcmcia_socket0: cs: memory probe 0x0c0000-0x0fffff:
[   14.844443]  excluding 0xc0000-0xd7fff 0xe0000-0xfffff
[   14.844458] pcmcia_socket pcmcia_socket0: cs: memory probe 0xa0000000-0xa0ffffff:
[   14.844475]  excluding 0xa0000000-0xa0ffffff
[   14.844490] pcmcia_socket pcmcia_socket0: cs: memory probe 0x60000000-0x60ffffff:
[   14.844506]  excluding 0x60000000-0x60ffffff
[   14.879952] iwlwifi 0000:0c:00.0: Firmware has old API version, expected v5, got v2.
[   14.879954] iwlwifi 0000:0c:00.0: New firmware can be obtained from http://www.intellinuxwireless.org/.
[   14.880093] iwlwifi 0000:0c:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[   15.088292] iwlwifi 0000:0c:00.0: CONFIG_IWLWIFI_DEBUG disabled
[   15.088293] iwlwifi 0000:0c:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[   15.088295] iwlwifi 0000:0c:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[   15.088297] iwlwifi 0000:0c:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[   15.088369] iwlwifi 0000:0c:00.0: L1 Enabled - LTR Disabled
[   15.253825] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[   15.327104] cfg80211: World regulatory domain updated:
[   15.327105] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[   15.327107] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
[... 15 lines cut ...]
[ 2773.898478] iwlwifi 0000:0c:00.0: request for firmware file 'iwlwifi-5000-5.ucode' failed.
[ 2773.898494] iwlwifi 0000:0c:00.0: Direct firmware load failed with error -2
[ 2773.898496] iwlwifi 0000:0c:00.0: Falling back to user helper
[ 2773.901041] iwlwifi 0000:0c:00.0: request for firmware file 'iwlwifi-5000-4.ucode' failed.
[ 2773.901058] iwlwifi 0000:0c:00.0: Direct firmware load failed with error -2
[ 2773.901060] iwlwifi 0000:0c:00.0: Falling back to user helper
[ 2773.903397] iwlwifi 0000:0c:00.0: request for firmware file 'iwlwifi-5000-3.ucode' failed.
[ 2773.903631] iwlwifi 0000:0c:00.0: Firmware has old API version, expected v5, got v2.
[ 2773.903634] iwlwifi 0000:0c:00.0: New firmware can be obtained from http://www.intellinuxwireless.org/.
[ 2773.903761] iwlwifi 0000:0c:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[ 2773.928265] iwlwifi 0000:0c:00.0: CONFIG_IWLWIFI_DEBUG disabled
[ 2773.928269] iwlwifi 0000:0c:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[ 2773.928272] iwlwifi 0000:0c:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[... cut ...]
[ 2779.727085] iwlwifi 0000:0c:00.0: L1 Enabled - LTR Disabled
[ 2779.727547] iwlwifi 0000:0c:00.0: Radio type=0x0-0x2-0x0
[ 2779.865084] iwlwifi 0000:0c:00.0: L1 Enabled - LTR Disabled
[ 2779.865539] iwlwifi 0000:0c:00.0: Radio type=0x0-0x2-0x0
[ 2779.898578] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2796.251154] wlan0: authenticate with 20:a9:d3:7c:39:18
[ 2796.255350] wlan0: send auth to 20:a9:d3:7c:39:18 (try 1/3)
[ 2796.264200] wlan0: authenticated
[ 2796.264237] wlan0: waiting for beacon from 20:a9:d3:7c:39:18

rt6n1, 2018-04-15:
Dhalperi, may I kindly ask you for assistance, as I am experiencing exactly the same problem:

milan@fuji:~$ uname -a
Linux fuji 3.13.0-24-generic #46-Ubuntu SMP Thu Apr 10 19:08:14 UTC 2014 i686 i686 i686 GNU/Linux
milan@fuji:~$ iw dev wlan2 link
Connected to ac:64:62:87:70:cc (on wlan2)
	SSID: RT6N1
	freq: 2412
	RX: 2054721 bytes (6335 packets)
	TX: 223094 bytes (1552 packets)
	signal: -22 dBm
	tx bitrate: 81.0 MBit/s MCS 4 40Mhz

	bss flags:	short-slot-time
	dtim period:	0
	beacon int:	100
milan@fuji:~$ iw dev wlan2 scan
command failed: Operation not permitted (-1)
milan@fuji:~$ sudo iw dev wlan2 scan
[sudo] password for milan: 
BSS ac:64:62:87:70:cc (on wlan2) -- associated
	TSF: 734622594 usec (0d, 00:12:14)
	freq: 2412
	beacon interval: 100
	capability: ESS ShortSlotTime APSD (0x0c01)
	signal: -22.00 dBm
	last seen: 104 ms ago
	Information elements from Probe Response frame:
	SSID: RT6N1
	Supported rates: 1.0* 2.0* 5.5* 11.0* 9.0 18.0 36.0 54.0 
	DS Parameter set: channel 1
	ERP: Barker_Preamble_Mode
	Extended supported rates: 6.0 12.0 24.0 48.0 
	HT capabilities:
		Capabilities: 0x101e
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			No RX STBC
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 4 usec (0x05)
		HT RX MCS rate indexes supported: 0-15, 32
		HT TX MCS rate indexes are undefined
	HT operation:
		 * primary channel: 1
		 * secondary channel offset: above
		 * STA channel width: any
		 * RIFS: 0
		 * HT protection: 20 MHz
		 * non-GF present: 1
		 * OBSS non-GF present: 0
		 * dual beacon: 0
		 * dual CTS protection: 0
		 * STBC beacon: 0
		 * L-SIG TXOP Prot: 0
		 * PCO active: 0
		 * PCO phase: 0
	WMM:	 * Parameter version 1
		 * u-APSD
		 * BE: CW 15-63, AIFSN 3
		 * BK: CW 15-1023, AIFSN 7
		 * VI: CW 7-15, AIFSN 2, TXOP 3008 usec
		 * VO: CW 3-7, AIFSN 2, TXOP 1504 usec
	Country: RS	Environment: Indoor/Outdoor
		Channels [1 - 13] @ 16 dBm
BSS 10:05:01:92:b4:54 (on wlan2)
	TSF: 599494482463 usec (6d, 22:31:34)
	freq: 2437
[... 289 lines cut ...]
[  247.664126] iwlwifi 0000:0a:00.0: Direct firmware load failed with error -2
[  247.664129] iwlwifi 0000:0a:00.0: Falling back to user helper
[  247.664731] iwlwifi 0000:0a:00.0: request for firmware file 'iwlwifi-5000-4.ucode' failed.
[  247.664754] iwlwifi 0000:0a:00.0: Direct firmware load failed with error -2
[  247.664756] iwlwifi 0000:0a:00.0: Falling back to user helper
[  247.665917] iwlwifi 0000:0a:00.0: request for firmware file 'iwlwifi-5000-3.ucode' failed.
[  247.684843] iwlwifi 0000:0a:00.0: Firmware has old API version, expected v5, got v2.
[  247.684850] iwlwifi 0000:0a:00.0: New firmware can be obtained from http://www.intellinuxwireless.org/.
[  247.684995] iwlwifi 0000:0a:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[  247.746541] iwlwifi 0000:0a:00.0: CONFIG_IWLWIFI_DEBUG disabled
[  247.746545] iwlwifi 0000:0a:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[  247.746547] iwlwifi 0000:0a:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[  247.746549] iwlwifi 0000:0a:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[  247.746653] iwlwifi 0000:0a:00.0: L1 Enabled; Disabling L0S
[  247.785970] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[  247.824023] systemd-udevd[2238]: renamed network interface wlan0 to wlan2
[  247.829206] iwlwifi 0000:0a:00.0: L1 Enabled; Disabling L0S
[  247.832524] iwlwifi 0000:0a:00.0: Radio type=0x0-0x2-0x0
[  247.972897] iwlwifi 0000:0a:00.0: L1 Enabled; Disabling L0S
[  247.976014] iwlwifi 0000:0a:00.0: Radio type=0x0-0x2-0x0
[  248.010787] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
[  248.011768] IPv6: ADDRCONF(NETDEV_UP): wlan2: link is not ready
[  254.651793] wlan2: authenticate with ac:64:62:87:70:cc
[  254.655090] wlan2: send auth to ac:64:62:87:70:cc (try 1/3)
[  254.657432] wlan2: authenticated

mj85eng, 2019-01-22:
I have the same problem
I have a DELL E6500 laptop with nic iwl 5300 running version 14.04.1 of the Linux kernel 3.13 on it
I've followed all the commands in the instructions
But despite the pinging of the Access Point, the csi.dat file is still empty
and when I boot into Ubuntu, I see the following error:
iwlwifi: module has bad taint, not creating trace events