dhalperi/linux-80211n-csitool-supplementary #132: error in monitor mode

star90, 2015-12-10:
Hi, 

I am trying to use monitor mode. However,  when I rry to run 'sudo ./random_packets 1 100 1', it says:
Generating packet payloads 
Initializing LORCON
Error opening LORCON interface.

What's the meaning of this?  I have done the following steps:
cd lorcon-old-master
./configure
make
sudo make install

Do you guys have some opinions about this? Looking forward to reply. 

Thanks a lot.

ozawali, 2015-12-17:
you don't set up the injection model so the command is not useful

zainalabidenakhtar, 2016-07-19:
@ozawali how to setup injection model? because i am also getting the same error, also when I run ./monitor_csi.sh 64 HT20 i get the following:
[image]