dhalperi/linux-80211n-csitool-supplementary #133: How can i improve the sampling rate?

Darwintam, 2015-12-10:
Now i use "ping <AP IP> -i 1/rate" , But this approach seems to have lost package. So i wanna try to use the injection ,i see the instructions in Readme is that:
"	
echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate`
	sudo ./random_packets 1 100 1
"
i wanna to make clear how to set my sampling rate in this way, and how to use the 'inject' ?
In this way ,does the sender also should have wl5300?
THANKYOU!!

ozawali, 2015-12-17:
you must set the injection model FAQ1

Darwintam, 2015-12-17:
How to use the injection mode? Hah I really do not know how to use it.

What  mean the 'echo 0x4101 | sudo tee find /sys -name monitor_tx_rate
sudo ./random_packets 1 100 1' is?    

sorry  i am a rookie  ，so i have lots of problems ha-ha

lixiaohai, 2015-12-25:
you need to read the commands.h clearly.
https://github.com/dhalperi/linux-80211n-csitool/blob/csitool-3.13/drivers/net/wireless/iwlwifi/dvm/commands.h#L245-L334