dhalperi/linux-80211n-csitool-supplementary #135: Unable to connect to any AP after reloading the driver.

hot-dog, 2015-12-20:
Hi.
I have some problems when using 80211n csitool. My operating system is Ubuntu 14.04 and my AP is a TP-LINK wireless router, I set it to '802.11n-only' and it's not encrypted. I installed the driver following the new Installation Instructions, I did all the steps and it seems the installation is successful because I didn't see any errors. But after I did step 5, I cannot connect to any APs. If I reboot, I can connect to my AP but I cannot get any csi data, maybe the CSI logging is disabled. However after I reload the driver, I cannot connect to my AP anymore. Any advice?

The output of 'uname -a':
Linux kexu-pc 3.19.0-25-generic #26~14.04.1-Ubuntu SMP Fri Jul 24 21:16:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

The output of 'dmesg | grep iwl':
[  549.268098] iwlwifi: module verification failed: signature and/or  required key missing - tainting kernel
[  549.270275] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  549.270551] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-5.ucode failed with error -2
[  549.284653] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-4.ucode failed with error -2
[  549.284664] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-3.ucode failed with error -2
[  549.304635] iwlwifi 0000:03:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[  549.355754] iwlagn: connector callback registered
[  549.355765] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[  549.355766] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[  549.355767] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[  549.355768] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[  549.355812] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[  549.402745] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[  549.447663] iwlwifi 0000:03:00.0 wlan1: renamed from wlan0
[  549.457269] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[  549.460201] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  549.596789] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[  549.599715] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  925.860653] iwlagn: connector callback deregistered
[  941.504546] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  941.505129] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-5.ucode failed with error -2
[  941.505153] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-4.ucode failed with error -2
[  941.505173] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-3.ucode failed with error -2
[  941.505560] iwlwifi 0000:03:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[  941.523143] iwlagn: connector callback registered
[  941.523158] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[  941.523159] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[  941.523160] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[  941.523162] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[  941.523209] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[  941.544889] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[  941.547365] iwlwifi 0000:03:00.0 wlan1: renamed from wlan0
[  941.563975] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[  941.566915] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[  941.703497] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[  941.706424] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[70575.017089] iwlagn: connector callback deregistered
[70605.520383] iwlwifi 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[70605.520537] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-5.ucode failed with error -2
[70605.520544] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-4.ucode failed with error -2
[70605.520549] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-5000-3.ucode failed with error -2
[70605.520704] iwlwifi 0000:03:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
[70605.570387] iwlagn: connector callback registered
[70605.570400] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[70605.570402] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[70605.570403] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[70605.570405] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[70605.570451] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[70605.591955] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[70605.740516] iwlwifi 0000:03:00.0 wlan1: renamed from wlan0
[70605.761470] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[70605.764403] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0
[70605.903272] iwlwifi 0000:03:00.0: L1 Disabled - LTR Disabled
[70605.906207] iwlwifi 0000:03:00.0: Radio type=0x0-0x2-0x0

hot-dog, 2015-12-21:
I have solved this problem. Yesterday, I reinstalled it and merged the driver modifications into the source tree of my kernel version which I did not do before. After the installation, I didn't reboot my computer. Today, I reboot my machine and I found I can connect to my AP after reloading the modified driver, the 'log_to_file' also worked. Thanks!

closed 2015-12-22 by dhalperi

tfsanam, 2017-04-07:
Hi, 

Can you please tell me how did you connect your AP with is utilities?