dhalperi/linux-80211n-csitool-supplementary #146: Installing the Modified Wireless Driver--- Pull is not possible

asifhanif1122, 2016-01-13:
[image]
[image]

I am trying to install modified wireless driver on Linux Ubuntu whose kenrel version is [ 3.19.0-15-generic ] and corresponding kernel tag is [  Ubuntu-3.19.0-15.15 ]. When i run following commands i receive an error on terminal saying that "fatal: remote ubuntu exists" after this command
 "git remote add ubuntu git://kernel.ubuntu.com/ubuntu/ubuntu-${DISTRIB_CODENAME}.git"  .
I ignore it and run next command on terminal "git pull --no-edit ubuntu ${UBUNTU_KERNEL_TAG}" but again receive an error that say that "Pull is not possible you have unmerged files". My colleague told me to try following commands to resolve this issue.

git fetch origin
git reset --hard origin/master
git pull  

Afterwards i again ran previous commands but to no avail. Please suggest me what to do to get these issues resolved. I have attached screenshots of errors that i encountered. 
[image]

1xiangyu, 2016-03-23:
Hi,@asifhanif1122 It seems i have the same problem. Do you solve it?

asifhanif1122, 2016-03-23:
@1xiangyu Which Ubuntu version are you using?

1xiangyu, 2016-03-23:
Ubuntu 4.2.0-27.32~14.04.1-generic 4.2.8-ckt1    
Ubuntu Kernel Tag Ubuntu-Its-4.2.0-27.32_14.04.1
Mainline Kernel Version 4.2.8-ckt1
Please forgive my verbosity, I'm new to linux. Thanks in advance for your help.

1xiangyu, 2016-03-23:
@asifhanif1122 Thanks in advance for your help.
Ubuntu 4.2.0-27.32~14.04.1-generic 4.2.8-ckt1
Ubuntu Kernel Tag Ubuntu-Its-4.2.0-27.32_14.04.1
Mainline Kernel Version 4.2.8-ckt1
Please forgive my verbosity, I'm new to linux.

asifhanif1122, 2016-03-23:
@1xiangyu  Try to use this version of ubuntu. "Ubuntu 14.04.2 LTS trusty" . Previously i was working on 15.04 version of ubuntu and it created all mess that i posted above in the form of screen shots. If problem persists, i will send you my files for installation of modified drivers.

1xiangyu, 2016-03-24:
@asifhanif1122 thanks for your prompt reply. Let me try.

qincaixj, 2016-03-31:
@asifhanif1122  i was stuck at the same point with you, then i change my ubantu version to "Ubuntu 14.04.2 LTS trusty".but when I want to unload the drive by input"sudo modprobe -r iwlwifi mac80211",it showed "

tara@ubuntu:~$ sudo modprobe -r iwldvm iwlwifi mac80211
rmmod: ERROR: missing module name.
modprobe: FATAL: Error running remove command for iwlwifi

what do you think might be the problem?I'm a green hand in Linux,Thanks a lot.

MBrend15, 2016-04-19:
I struggled with a similar issue for the last couple of days. I am currently using wily, kernel Ubuntu-4.2.0-16.19. At the step below: 

CSITOOL_KERNEL_TAG=csitool-$(uname -r | cut -d . -f 1-2)
git clone https://github.com/dhalperi/linux-80211n-csitool.git
cd linux-80211n-csitool
git checkout ${CSITOOL_KERNEL_TAG}

I checked out the version with tag 4.1. After that was loaded into the CIStool repository, I then executed git pull -x theirs --no-edit ubuntu 4.2.0-16.19, to pull for my kernel and automatically merge the changes that I pulled. I was able to work through the rest of the installation without any issues. 

I hope this helps, sorry if it's too late.

dpward, 2016-05-24:
I've been temporarily without laptops on which I can do any testing. Prior to that I had merged the driver modifications with each kernel release through Linux 4.1. However as you mentioned Ubuntu 14.04.4 is now based on Linux 4.2, and Ubuntu does not support old HWE releases for very long. I was able to merge the driver modifications against Linux 4.2 without any conflicts, and another user tested that this is in fact providing CSI, so I've pushed it to the repository. There is now a csitool-4.2 tag so you should not have this problem in the future.

For Linux 4.3 and later (including Ubuntu 16.04) I need to do a manual merge with the changes (issue #129 ).

closed 2016-05-24 by dpward

Bashima, 2017-09-14:
Hi, I am trying to install the Linux CSI-tool in Ubuntu 16.04  Kernel 4.4.0-93-generic. Is it possible to do so?

vpeddint, 2017-09-21:
No, it doesn't work on Ubuntu 16.04. I tried, but failed to install.