dhalperi/linux-80211n-csitool-supplementary #151: 5000 packets received at Receiver.. Each packet contains information of 30 sub-carriers. Which packet number and the sub-carrier in that packet should be focused to observe changes in environment?

asifhanif1122, 2016-01-26:
Hi ,
I am working on a project in which our objective is to observe change in environment like presence of metal in bag or any thing related to this scenario. I send multiple number of packets from transmitter (Intel' 5300 wifi card using only one antenna) and receive them on 5300 wifi card connected to another laptop in Monitor mode. I want to process these packets on Matlab. But I am confused that which range of packets i.e ( from packet number 3000 to packet number 4000 or any other range ) should be processed? Second question is that every packet contains information i.e CSI of 30 sub-carriers but which of the sub-carriers i.e ( sub-carrier-1 or sub-carrier-20) should be used to observe changes in environment. I will be thankful for the help I am seeking for.

egaebel, 2016-04-21:
This sounds like an entire research problem, not something that can be answered on an issues page?

adi237, 2017-11-22:
@asifhanif1122 - Did you figure it out? If yes, would you mind sharing your solution?